kyleisah/Klipper-Adaptive-Meshing-Purging #2: Error when using adaptive bed meshing with exclusion zones

allenrowand, 2022-11-22:
I've been running exclusion zones on my bed mesh to probe my MRW magbed. Adaptive meshing errors and shuts down the MCU when the exclusion zones are enabled, commenting the regions out works as expected. I've attached the section of the error log where this occurred, let me know if I can provide anything else.

kyleisah, 2022-11-23:
I think the best thing I can do about this is extend the callout about beds with strong magnets to mention exclusion zones. I'm not terribly familiar with exclusion zones (I just use a regular magnetic sheet on my machines) but this does seem like something that could frustrate someone after all the setup is done. I will mention this more in the documentation. 😅

kyleisah, 2022-11-23:
Currently I don't know how to make adaptive meshing work around faulty regions. I believe Klipper usually probes a box around these areas and calculates a point from that. Getting both to work together correctly currently is difficult. However, I added a note to the Readme encouraging the submittal of a PR to make it work. Thanks for bringing this to my attention, as I hadn't even considered faulty regions when we were creating the project. I will leave this issue open and tagged as an enhancement!

dougshell, 2023-04-29:
Has there been any progress on this front?

kyleisah, 2023-05-05:
@dougshell none at all, sorry. I don't use exclusion zones so I cannot test it

closed 2023-07-06 by kyleisah

AudreyAP, 2023-07-08:
Wait, was this fixed?

kyleisah, 2023-07-08:
> Wait, was this fixed?

No, it wasn't. This kind of thing is out of the scope of KAMP, I don't know how to get it to work with exclusion zones without a ton of work and risk. I expect this to be much easier if Klipper had adaptive bed meshing within the bed_mesh.py module.

Hopefully, adaptive meshing will one day be a part of mainline klipper. I have a great mind helping to make this a reality.

kyleisah, 2023-07-08:
@voidtrance I know someone who uses exclusion zones that would be able to test the mainline version. @allenrowand would you be interested in trying it?

AudreyAP, 2023-07-08:
You can just add a couple of random coordinates if you want to test - doesn't need to actually have anything under it. I can write up a realistic one if you give me your bed dimensions :)

kyleisah, 2023-07-08:
@AudreyAP I've got a 250x250 bed, gimme a quick config and I'll try it

AudreyAP, 2023-07-08:
@kyleisah You can just use a generic MK3 bed config - here's a common one. Just goes in the bed mesh config block. 

```
faulty_region_1_min: 100.583, 187.376
faulty_region_1_max: 122.083, 229.376
faulty_region_2_min: 125.672, -18.124
faulty_region_2_max: 147.172, 23.876
faulty_region_3_min: 192.261, 5.126
faulty_region_3_max: 234.261, 26.626
faulty_region_4_min: 114.422, 39.876
faulty_region_4_max: 135.922, 81.876
faulty_region_5_min: 21.422, 87.126
faulty_region_5_max: 42.922, 129.126
faulty_region_6_min: 54.172, 97.376
faulty_region_6_max: 96.172, 118.876
faulty_region_7_min: 154.172, 97.376
faulty_region_7_max: 196.172, 118.876
faulty_region_8_min: 205.136, 87.126
faulty_region_8_max: 226.636, 129.126
faulty_region_9_min: 114.422, 134.376
faulty_region_9_max: 135.922, 176.376
faulty_region_10_min: 176.177, 191.394
faulty_region_10_max: 218.177, 212.894
```

kyleisah, 2023-07-08:
Well I'll be dipped, yeah, it totally works with exclusion zones when using the mainline version 😄

kyleisah, 2023-07-08:
[image]

@voidtrance is a literal wizard

AudreyAP, 2023-07-08:
This is fantastic. Thank you so much for getting this working!

dougshell, 2023-07-08:
I remember enabling kamp with faulty regions and getting errors.

Are you suggesting that I resemble and test?

AudreyAP, 2023-07-08:
Seems to be working @dougshell - above is a stock MK3 mesh.

dougshell, 2023-07-08:
I am certain I had errors before but good to know it got fixed.

Are you running the most up to date version?

I'll test with the old version tomorrow morning and report back if you
already updated

AudreyAP, 2023-07-08:
Kyle himself tested it above w/ mainline (current) KAMP.

kyleisah, 2023-07-08:
> I remember enabling kamp with faulty regions and getting errors. Are you suggesting that I resemble and test?
> […](#)

Yeah, this is not apart of KAMP though, this is a part of adaptive meshing being brought to mainline klipper, which Voidtrance is spearheading. Hopefully soon we can get it merged, and adaptive meshing will become another awesome base feature of klipper!

AudreyAP, 2023-07-08:
Ah. Misunderstood. 

Is it public anywhere yet?

kyleisah, 2023-07-08:
> Ah. Misunderstood.
> 
> Is it public anywhere yet?

Yes, *but* I'm going to leave that up to the maintainer of the fork if they want to expand the circle of testers right now. We've got a channel in the Voron Discord currently, and that's been serving as a decent spot to record and work through issues. If you're there, you're more than welcome to take a look and try it out.

dougshell, 2023-07-09:
including faulty_regions works with the KAMP prior to the latest update with the most recent publicly available klipper modules.

AudreyAP, 2023-07-09:
@dougshell just to confirm - by publicly available klipper modules do you mean official klipper from the klipper repo? Or do you mean the WIP adaptive mesh stuff?

dougshell, 2023-07-09:
I only have standard klipper from the repo