kyleisah/Klipper-Adaptive-Meshing-Purging #3: Adaptive purge trigger max_extrude_cross_section error

allenrowand, 2022-11-22:
When I tried to run the adaptive purge my print stopped and Mainsail warned that I exceeded the max extrude cross section; 4.1 was requested, the limit is .64. I was able to get purge working by adding max_extrude_cross_section: 5 to my config.

If the value needs to be bumped up you might want to call it out in the docs. Thanks for the code, it's pretty cool!

[image]

kyleisah, 2022-11-23:
[image]

Ah yeah haha, that'll happen. It is called out in the config file, but I think adding it somewhere in the main Readme would be beneficial.

allenrowand, 2022-11-23:
I looked straight at it but didn't see it. Doh! Sorry.

kyleisah, 2022-11-23:
Added some notes to the readme. Thank you!

closed 2022-11-23 by kyleisah