kyleisah/Klipper-Adaptive-Meshing-Purging #4: adaptive mash not adapting

shirroid, 2022-11-26:
I am trying to run the code and it's working fine but it doesn't adapt to what I am printing. It meshes the whole bed even when I only print in the needle. The same is true for adaptive purge too.
exclude objects is enabled in the config and the slicer.

kyleisah, 2022-11-26:
Hey there! Are you getting a readout saying something like "0 points, clamping to bed mesh"? There is a fix for this located at the bottom of the repo under Troubleshooting.

Cliff-333, 2022-11-26:
I was about to comment on this to say I was also having this issue but after adding the M117 to the beginning of my start gcode, it is working. I wasn't originally looking for the error in the console because I (incorrectly) thought the error would cancel the print. Thanks for your work on this, @kyleisah.

kyleisah, 2022-11-27:
@Cliff-333 can you please upload your Klipper log file here?

shirroid, 2022-11-27:
I succeeded in resolving the issue thank you very much

Cliff-333, 2022-11-27:
Here is my klipper log file
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/10099105/klippy.log)

kyleisah, 2022-11-27:
@Cliff-333 Hmmm, this all seems to be okay. Can you upload the gcode file that was giving you issues on the 2.4?

Cliff-333, 2022-11-27:
Sure. Here is the gcode. Github wouldn't let me upload it as a .gcode file so I had to zip it.
[Cali-Dragon_v1.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/10099148/Cali-Dragon_v1.zip)

kyleisah, 2022-11-27:
[image]

Seems like for whatever reason, exclude_object is not processing your gcode files on your 2.4. This happened to me before. I had to pretty much nuke my moonraker installation and reinstall it to fix it. Before you do that though, double check you're labeling objects in the slicer and also have `[exclude_object]` in your printer.cfg and have the gcode processing bit in your moonraker.conf file. Let me know what you find!

Cliff-333, 2022-11-27:
Yeah, I have all of those prerequisites. I guess I'll try reinstalling moonraker and will report back.

[image]
[image]
[image]

kyleisah, 2022-11-27:
Once you reinstall moonraker, you'll have to reupload the gcode to confirm, it doesn't process them recursively unfortunately

Cliff-333, 2022-11-27:
Reinstalling moonraker did the trick. Adaptive mesh and purge are now working. Thanks so much for your help!

closed 2022-11-29 by kyleisah