kyleisah/Klipper-Adaptive-Meshing-Purging #5: Can't get it to work

110Tim706, 2022-11-27:
So what i did:
- downloaded both cfg files
- uploaded them to the config section
- included them in printer.cfg 
- pasted:

[file_manager]
enable_object_processing: True

in moonraker.conf
- pasted [exclude_object] in printer.cfg

still not working. What is meant with this step?
Double check your PRINT_START macro to ensure BED_MESH_CALIBRATE and/or ADAPTIVE_PURGE macros are defined.
haven't got these as macros.

110Tim706, 2022-11-27:
what needs to be  in print_start?

110Tim706, 2022-11-27:
Had to configure the macro.

[gcode_macro PRINT_START]
gcode:
  BED-MESH-CALIBRATE

closed 2022-11-27 by 110Tim706

kyleisah, 2022-11-27:
@110Tim706 

Oh I see, so it sounds like you've usually just used the slicer's start gcode section instead of having a print_start macro in Klipper?

Having a Print_Start macro is honestly ideal because then you can expect the same behavior from your machine regardless of what slicer you use. 😄 

Here's this link which will give you some more information:
https://www.klipper3d.org/Slicers.html#klipper-gcode_macro