kyleisah/Klipper-Adaptive-Meshing-Purging #6: Add Adaptive Probe Count

ChaosBlades, 2022-11-27:
When printing parts with a very large first layer I run a 9x9 mesh due to a hilly bed but this is massive overkill for printing a tiny part. If the part is small enough even 1 point will do. So it would bring this macro to the next level if it could set a minimum and maximum probe count. There is no point in meshing 9x9 just to print a benchy on a 350mm bed for example. A simple 3x3 is more than enough

I think this macro has this functionality.
https://gist.github.com/ChipCE/95fdbd3c2f3a064397f9610f915f7d02

kyleisah, 2022-11-27:
As it is right now, whatever you have set as your `probe_count` under `[bed_mesh]` in `printer.cfg` will be the maximum probe count, and KAMP will adaptively scale it. Typically it will make a 3x3 as the smallest, or even smaller depending.

[image]

You can see here how the maximum is determined in Adaptive_Mesh.cfg. 

I don't think having a minimum makes sense, because the point is to have a small and rich mesh rather than a large and wasteful mesh.

kyleisah, 2022-11-27:
Also, a 1 point mesh isn't really a mesh. You could just home Z to the center of the print area instead and not run a mesh, but I tend to think most people who don't use a bed mesh wouldn't want an adaptive mesh

ChaosBlades, 2022-11-28:
Your right, I didn't notice it actually does adjust the mesh size. 

Should have read more closely.

> KAMP made a mesh that is 7x4

closed 2022-11-28 by ChaosBlades

Kosh42, 2022-12-05:
I came here to request this exact feature. Excellent that it is already in place.

I may try and create a pull request later to make it more explicit in the docs so people don't raise it in future.

KAMP is an awesome invention. Thanks for sharing your hard work with the community.