kyleisah/Klipper-Adaptive-Meshing-Purging #10: console reports one point less, this should be 9 points in 3x3 grid

vajonam, 2022-11-29:
suspect the `| count` is returned based on 1 index and no 0.

https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/blob/73fcfbf149e3db56f0d6fd0f8ed09f9575589900/Configuration/Adaptive_Mesh.cfg#L40

```
10:57 AM 8 points, clamping to mesh [(5.0, 5.0) (201.0, 210.0)]
10:57 AM Object bounds, clamped to the bed_mesh: (78.44, 55.81), (153.69, 174.16)
10:57 AM Algorithm: lagrange
10:57 AM Points: x: 3, y: 3
10:57 AM Generating new points...
10:57 AM bed_mesh: generated points
Index | Tool Adjusted | Probe
10:57 AM 0 | (122.4, 64.8) | (78.4, 55.8)
10:57 AM 1 | (160.1, 64.8) | (116.1, 55.8)
10:57 AM 2 | (197.7, 64.8) | (153.7, 55.8)
10:57 AM 3 | (197.7, 124.0) | (153.7, 115.0)
10:57 AM 4 | (160.1, 124.0) | (116.1, 115.0)
10:57 AM 5 | (122.4, 124.0) | (78.4, 115.0)
10:57 AM 6 | (122.4, 183.2) | (78.4, 174.2)
10:57 AM 7 | (160.1, 183.2) | (116.1, 174.2)
10:57 AM 8 | (197.7, 183.2) | (153.7, 174.2)```
```

kyleisah, 2022-11-29:
The points are counted in binary, and start at 0.

vajonam, 2022-11-29:
counts that presented to users in the console  usually start at 1. you can close this issue not a biggie for me.

kyleisah, 2022-11-29:
Sorry, I misunderstood, that does seem to be incorrect

kyleisah, 2022-12-11:
@vajonam Hey, so after some confirmation. the console is reporting how many points are being defined by exclude_object. when it says `8 points, clamping to bed mesh`, it is because exclude_object determined 8 points of object bounds

kyleisah, 2022-12-11:
Suppose that should probably be changed to avoid confusion 😅

kyleisah, 2022-12-14:
I updated the nomenclature for that in the macros. Thanks for reaching out!

closed 2022-12-14 by kyleisah