kyleisah/Klipper-Adaptive-Meshing-Purging #8: Issue when using the adaptive mesh macro

celtare21, 2022-11-29:
Hey guys! Awesome work. I've wanted to try it too but when including the mesh macro i get the following:

`mcu_temperature 'mcu' nominal base=-279.171598 slope=1332.692308
Configured MCU 'mcu' (1024 moves)
Configured MCU 'rpi' (1024 moves)
mcu 'rpi': got {u'query_ticks': 1568, u'buffered': 0, u'limit_count': 0, '#receive_time': 924969.225715379, u'clock': 3222697521L, u'oid': 1, u'next_sequence': 0, u'fifo': 0, '#name': u'adxl345_status', '#sent_time': 924969.225545434}
Starting heater checks for heater_bed
bed_mesh: generated points
Index |  Tool Adjusted  |   Probe
  0   | (30.0, 64.0)    | (30.0, 26.0)
  1   | (58.3, 64.0)    | (58.3, 26.0)
  2   | (86.7, 64.0)    | (86.7, 26.0)
  3   | (115.0, 64.0)   | (115.0, 26.0)
  4   | (143.3, 64.0)   | (143.3, 26.0)
  5   | (171.6, 64.0)   | (171.6, 26.0)
  6   | (200.0, 64.0)   | (200.0, 26.0)
  7   | (200.0, 90.0)   | (200.0, 52.0)
  8   | (171.6, 90.0)   | (171.6, 52.0)
  9   | (143.3, 90.0)   | (143.3, 52.0)
  10  | (115.0, 90.0)   | (115.0, 52.0)
  11  | (86.7, 90.0)    | (86.7, 52.0)
  12  | (58.3, 90.0)    | (58.3, 52.0)
  13  | (30.0, 90.0)    | (30.0, 52.0)
  14  | (30.0, 116.0)   | (30.0, 78.0)
  15  | (58.3, 116.0)   | (58.3, 78.0)
  16  | (86.7, 116.0)   | (86.7, 78.0)
  17  | (115.0, 116.0)  | (115.0, 78.0)
  18  | (143.3, 116.0)  | (143.3, 78.0)
  19  | (171.6, 116.0)  | (171.6, 78.0)
  20  | (200.0, 116.0)  | (200.0, 78.0)
  21  | (200.0, 142.0)  | (200.0, 104.0)
  22  | (171.6, 142.0)  | (171.6, 104.0)
  23  | (143.3, 142.0)  | (143.3, 104.0)
  24  | (115.0, 142.0)  | (115.0, 104.0)
  25  | (86.7, 142.0)   | (86.7, 104.0)
  26  | (58.3, 142.0)   | (58.3, 104.0)
  27  | (30.0, 142.0)   | (30.0, 104.0)
  28  | (30.0, 168.0)   | (30.0, 130.0)
  29  | (58.3, 168.0)   | (58.3, 130.0)
  30  | (86.7, 168.0)   | (86.7, 130.0)
  31  | (115.0, 168.0)  | (115.0, 130.0)
  32  | (143.3, 168.0)  | (143.3, 130.0)
  33  | (171.6, 168.0)  | (171.6, 130.0)
  34  | (200.0, 168.0)  | (200.0, 130.0)
  35  | (200.0, 194.0)  | (200.0, 156.0)
  36  | (171.6, 194.0)  | (171.6, 156.0)
  37  | (143.3, 194.0)  | (143.3, 156.0)
  38  | (115.0, 194.0)  | (115.0, 156.0)
  39  | (86.7, 194.0)   | (86.7, 156.0)
  40  | (58.3, 194.0)   | (58.3, 156.0)
  41  | (30.0, 194.0)   | (30.0, 156.0)
  42  | (30.0, 220.0)   | (30.0, 182.0)
  43  | (58.3, 220.0)   | (58.3, 182.0)
  44  | (86.7, 220.0)   | (86.7, 182.0)
  45  | (115.0, 220.0)  | (115.0, 182.0)
  46  | (143.3, 220.0)  | (143.3, 182.0)
  47  | (171.6, 220.0)  | (171.6, 182.0)
  48  | (200.0, 220.0)  | (200.0, 182.0)
Starting heater checks for extruder
webhooks: registering remote method 'shutdown_machine' for connection id: 3045110248
webhooks: registering remote method 'reboot_machine' for connection id: 3045110248
webhooks: registering remote method 'pause_job_queue' for connection id: 3045110248
webhooks: registering remote method 'start_job_queue' for connection id: 3045110248
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: tuple object has no element 1
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "<template>", line 6, in top-level template code
UndefinedError: tuple object has no element 1
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: tuple object has no element 1
`

kyleisah, 2022-11-29:
Hello, I'm not sure why you're getting this issue. I haven't seen this come up before. Are you running the latest version of klipper and moonraker?

kyleisah, 2022-11-29:
@celtare21 seems like you only have one number defined in `probe_count` and it needs to be a number for both min and max. So instead of `7`, you need to have `7,7`

closed 2022-11-29 by kyleisah

celtare21, 2022-12-01:
> @celtare21 seems like you only have one number defined in `probe_count` and it needs to be a number for both min and max. So instead of `7`, you need to have `7,7`

I can confirm this was the issue. Thank you!