kyleisah/Klipper-Adaptive-Meshing-Purging #9: repo marked dirty when using  moonraker update manager

vajonam, 2022-11-29:
firstly great set of macros! 

i am using the moonraker update maanger

```
[update_manager kamp]
type: git_repo
primary_branch: main
path: ~/kamp
origin: https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging.git
```

however the repo is marked dirty because of my edits. is it feasable to add a config macro that sets up the variables inside the main macros. so those can remain clean and will cleanly update from github? 

```
[gcode_macro SETUP_KAMP]
gcode:
    SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=variable_fuzz_enable VALUE=True
    # setup other variables
```

will this work? I can try it out, but not sure if the variables get reset what are you thoughts?

kyleisah, 2022-11-29:
Hello, we do not currently support moonraker updating for this reason. Any changes to configs would mark it dirty, and there's not really any scripts or anything that would need to be maintained. This has come up before though, so I will look into how we can handle this.

vajonam, 2022-11-29:
the moonraker timelapse macro uses a SETUP macro similar to above that is generated by mainsail and UI configuration for the timelapse. This is then called, so instead of editing the config files. allows us to keep the gitub as well custom configs per user. 

```
# Use _SET_TIMELAPSE_SETUP [ENABLE=value] [VERBOSE=value] [PARK_ENABLE=value] [PARK_POS=value] #
```

kyleisah, 2022-12-11:
I have someone who may be interested in getting this to work correctly. Let me see what they come up with!

kyleisah, 2022-12-11:
@jtrmal would you be willing to take a look? I know this is something you wanted to have working.

jtrmal, 2022-12-11:
Gladly!

jtrmal, 2023-02-20:
PR #32  should resolve this

vajonam, 2023-02-21:
@jtrmal  just tried it. Trying with my klicky probe seems to fail 

```
SETUP_KAMP_MESHING DISPLAY_PARAMETERS=1 LED_ENABLE=1 FUZZ_ENABLE=1  PROBE_DOCK_ENABLE=1
SETUP_VORON_PURGE DISPLAY_PARAMETERS=1 ADAPTIVE_ENABLE=1
```

```
12:13 PM
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: 'float' object cannot be interpreted as an integer
12:13 PM
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: 'float' object cannot be interpreted as an integer
12:13 PM 16 object points, clamping to bed mesh [(40.0, 40.0) (260.0, 260.0)]
12:13 PM detach_macro: 'Dock_Probe'
12:13 PM attach_macro: 'Attach_Probe'
12:13 PM probe_dock_enable: 1
12:13 PM fuzz_max : 4.000000
12:13 PM fuzz_min : 0.000000
12:13 PM fuzz_enable : 1
12:13 PM status_macro: 'status_meshing'
12:13 PM led_enable : 1
```

removing the setup macro  and hardcoding seems to fix it. looks like something is float

vajonam, 2023-02-21:
removing FUZZ_ENABLE=1 from my setup call seems to resolve it. something going on there.

jtrmal, 2023-02-21:
interesting, I will try to figure it out and fix it today
y.

jtrmal, 2023-02-21:
I don';t see anything wrong ATM, but I will try to debug it after work
y.

vajonam, 2023-02-21:
or this

```
-        {% set fuzz_range = range(fuzz_min * 100 | int, fuzz_max * 100 | int + 1) %}
+        {% set fuzz_range = range((fuzz_min * 100) | int, (fuzz_max * 100) | int + 1) %}

```

jtrmal, 2023-02-21:
I think the second path is more correct -- the range itself should be allowed to be float (millimeters)

jtrmal, 2023-02-21:
do you wanna create PR? I can do it, if you don't

vajonam, 2023-02-21:
which is the right way to do it? the parenthesis around the multiplication? or the forcing it to int on variable parsing?

vajonam, 2023-02-21:
Yeah I can. standby

jtrmal, 2023-02-21:
I think the second way (parentheses) and downcast to int

kyleisah, 2023-02-21:
Thanks for the fine work dudes!

closed 2023-02-21 by kyleisah

vajonam, 2023-02-22:
just tested to update to my own PR :-) nice! this will drive more easier adoption