kyleisah/Klipper-Adaptive-Meshing-Purging #11: Invalid speed in 'G1 F0'

blonestar, 2022-11-30:
I was really happy to try adaptive meshing but it stops immediately after enormous purging with err message "Invalid speed in 'G1 F0'", so not sure what to try next.

Also is it really needed that much of purge?
Had to set max_extrude_cross_section to 5 as I was getting errror:
"Move exceeds maximum extrusion (4.165mm^2 vs 1.440mm^2)"
Some default value there should be for a reason, I guess.

kyleisah, 2022-11-30:
These parameters are all easily adjusted inside of the adaptive purge cfg file, which also mentions the need for adjusting the max extrude cross section to 5. As far as the G1 F0 command goes, that is something your slicer is doing.

blonestar, 2022-11-30:
Error solved in slicer, will play with adaptive purge parameters. :)
Thank you for a nice work!

closed 2022-11-30 by blonestar