kyleisah/Klipper-Adaptive-Meshing-Purging #13: Does not print object on the same spot as it measure and no purgeline

jeor112, 2022-12-03:
Great work. thanks!

New to this and it's probably my bad, but it does not make the purgeline and it do not measure at the same place as where the object is  been printed on the bed.

I'm using BLTouch.

I have removed the purgeline settings from the start_print macro. Is that correct?

Please can you point me in the right direction?

I have followed the guide, but I must have overlooked something.

Thanks in advance.

This is my start gcode in Klipper:
[gcode_macro START_PRINT_ADAPTIVE_MESH]
gcode:
    G90
    G28
    BED_MESH_CLEAR
    Z_TILT_ADJUST
    BED_MESH_CALIBRATE

jeor112, 2022-12-05:
UPDATE
I found that I have made an error in the Adaptive_Purge.cfg, so that part is working now.
But, I still have the issue, that it is not measureing at the same place as where the print is being done.

Hopefully someone would be so kind to help me figuring that out.

Thanks

kyleisah, 2022-12-11:
@jeor112 Typically when that happens, it's because you have either not defined your [probe offsets](https://www.klipper3d.org/Probe_Calibrate.html#calibrating-probe-x-and-y-offsets) correctly, have not defined your bed's origin correctly, or have some sort of origin offset in your slicer. Do some work to make sure that your printer is locating itself where it is expecting to be.

jeor112, 2022-12-12:
Thanks for your answer! You are right, it was my probe offset. 🤬 Somehow it was set to negative, so correcting that, everything works perfect! Really nice feature to use for a big printer bed, so that I'll be using it from now on 👍🙂

kyleisah, 2022-12-14:
Hey that's awesome! I'm so glad you got that fixed up! 😄

closed 2022-12-14 by kyleisah