kyleisah/Klipper-Adaptive-Meshing-Purging #15: bed mesh only on whole bed

Kevincondor, 2022-12-07:
hello,i'm having issues,where the "adaptive mshing" is done on all the bed,even with a 20x20 size object.
cura/klipper/fluidd last update

 0 points, clamping to mesh [(25.0, 35.0) (225.0, 220.0)]
Object bounds, clamped to the bed_mesh: (25.0, 35.0), (225.0, 220.0)

i do maybe need to create a default mesh before?

kyleisah, 2022-12-11:
@Kevincondor Yeah, the M82/M83 debacle is something that has came up  a lot since release. I'm wondering if adding M83 at the end of the macro will fix it for everyone. I'm going to check some gcode.

kyleisah, 2022-12-11:
It looks like if you use relative extrusion in your slicer, after the purge macro is called, the slicer will set whatever extrusion mode it uses. So I think adding M82 at the end would fix the errors for everyone. :)

Kevincondor, 2022-12-11:
i'll try it too tomorrow evening after work and let you know :)

kyleisah, 2022-12-11:
I'm fairly certain it will work, but please do! I am currently working on the repo a bit to make some small improvements and fulfill some requests.

Kevincondor, 2022-12-12:
yea, M82 at the end of adaptive purge seems to work fine :) now i need to figure out how to set the amounts XD any tips for a revo voron stealthburner?

kyleisah, 2022-12-14:
Glad to hear! I actually don't have a Revo, so I couldn't tell you. Best thing to adjust first is the tip_distance parameter. Once you have that dialed in you should be all set from there.

closed 2022-12-14 by kyleisah