kyleisah/Klipper-Adaptive-Meshing-Purging #17: object labeling

disneytoy, 2022-12-15:
Sorry to ask here,

you wrote: You must have object labeling enabled in your slicer. 

I am looking in the latest Cura and can not find object labeling. Is this compatible with Cura or just PrusaSlicer?

Is there a discord or place to ask questions or get help? I'm a newbee. I've got as far as: 

Double check your PRINT_START macro to ensure BED_MESH_CALIBRATE and/or ADAPTIVE_PURGE macros are defined.

they aren't but I can research that on my own. 

Thanks.

MAX
'

disneytoy, 2022-12-17:
UPDATE: I only got up to this part "**Double check your PRINT_START macro to ensure BED_MESH_CALIBRATE and/or ADAPTIVE_PURGE macros are defined"**  Don't think I set that up but on a test print today, It did infact only probe the print area. It did a regular purge line which I guess is fine.

I don't think if you have a single object in Cura, the "OBJECT LABELING" is not required. 

thanks

kyleisah, 2022-12-17:
Hey there! I meant to get back to you, I do apologize. Yes, it seems cura already does this natively. No need to enable labeling, it already takes care of it. As far as the PRINT_START stuff goes, a lot of folks use PRINT_START as a macro and define whatever they want inside of it that they want the machine to do before a print starts. This makes it really easy to have the same print startup across different slicers. If you'd like to try the adaptive purging, in your start-gcode section of cura, you'd have to add the command ADAPTIVE_PURGE at the end. This might make you have two different purge lines though. 

I am happy to hear the adaptive meshing is working! I am currently working on a way for users to make their own purge gcode and allow whatever purge they want to be adaptive. 😄

disneytoy, 2022-12-17:
Thanks for your response. You have done a great job. I thought it wouldn't work because I didn't complete the steps. But it did.  I generally print functional parts that are usually 2-4". So your Adaptive Meshing should save some time, but, I can't wait to see if I get much more accurate first layers. I've switched from mirror to a G10 bed. The G10 won't be as flat as the mirror, but, I am very impressed on its adhesion, and once it cools a bit the part is completely released on its own.

Thanks again!!

Max

closed 2022-12-17 by disneytoy