kyleisah/Klipper-Adaptive-Meshing-Purging #20: Adaptive mesh ignoring probe_count setting

dJOS1475, 2023-01-11:
Howdy, Adaptive mesh is ignoring my probe_count: 5,5 setting and only probing a 3x3 matrix. This setting was being properly implemented before adding Adaptive mesh to the mix.

[image]

[image]

PS, love your work team!

Cheers Derek

dJOS1475, 2023-01-11:
Hmmm, looks like it works fine for larger prints - is going to a 3x3 for a smaller print part of the "adaptive" nature of this system?

karlla1220, 2023-01-16:
Based on my reading of codes, it use probe_count only to define a step size between probing point. if given a step size is (30, 30) and print area is (60, 60) to (150, 150), then, the code would use 3 by 3 mesh. Thus, probe_count defines granularity of adaptive mesh. Seems reasonable.

dJOS1475, 2023-01-16:
Based on fine more testing, lager prints will use the full 5x5 matrix and smaller prints won’t. Which is fine, it’s just not clear in the doco.

closed 2023-01-16 by dJOS1475