kyleisah/Klipper-Adaptive-Meshing-Purging #21: Add option for traditional Line purge. PETG likes to get stuck to the nozzle on Voron Purge.

ChaosBlades, 2023-01-16:
The Adaptive Purge Voron Logo likes to get PETG stuck all over the nozzle then some point during the print deposits a large blob into the print.

Add an option for a traditional line purge at 0.4mm layer height for easy removal. Add options for adjustable length and vertical or horizontal purge. 

```
    ##Purge Line Gcode
    G92 E0;
    G90
    G0 X5 Y5 F6000
    G0 Z0.4
    G91
    G1 X120 E30 F1200;
    G1 Y1
    G1 X-100 E30 F1200;
    G92 E0;
    G90
```

kyleisah, 2023-01-20:
Hey there, sorry for not getting to you sooner. Having user-defined purges is planned, I just have to figure out a way to make it elegant. 

I like having the variables for the purge so you can easily change whatever params you want to adjust the purge to how you need it. 

I do agree, there should be a more generic purge line. I'll have to see what I can cook up.

kyleisah, 2023-02-04:
@ChaosBlades if I were to start making a generic purge, would you be able to test it?

ChaosBlades, 2023-02-04:
@kyleisah yeah, just need to fix my printer. Shouldn't be more than a few weeks.

kyleisah, 2023-02-19:
New purge has been added, go ahead and try it out!

closed 2023-02-19 by kyleisah