kyleisah/Klipper-Adaptive-Meshing-Purging #24: How to change corner for purge logo?

Juste1911, 2023-02-03:
I would like to start by saying that I really appreciate your work here, it's a masterpiece!

I've been using your code for about a month on my 2.4 & customized Ender 3s (couple of them)... First layers are just perfect now.

My issue is when using this code on a carriage that features a BlTouch or similar probes that have a retractable pin, basically my Enders both got upgraded with a Sprite Pro Extruder kit, the BlTouch when installed is positioned on the front left corner of that carriage.

To make this short, the BlTouch pin almost scrapes the purge logo everytime I print anything, and if something gets stuck in the purge, that's even worse,

I really like having the purge logo close to the print, I've tried increasing the distance and lowering the purge amount but I was wondering if something could be done to query for a safer corner like the rear right corner for example which is opposite to where the BlTouch is positioned.

Let's say my bed is 230 x 230... 0, 0 is the front left corner... 230, 230 would be the rear right corner,

Am I correct to assume that I should change x & y origin position in the x & y origin section to switch the corner where the purge happens or am I totally wrong? My conditional code skills are inexistent...

Thank you again for your hard work

```
{% if adaptive_enable == True %}
        {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}
        {% set x_origin = (all_points | map(attribute=0) | min | default(x_default + distance_to_object_x + size)) - distance_to_object_x - size %}
        {% set y_origin = (all_points | map(attribute=1) | min | default(y_default + distance_to_object_y + size)) - distance_to_object_y - size %}
        **{% set x_origin = ([x_origin, 0] | max) %}
        {% set y_origin = ([y_origin, 0] | max) %}**
    {% else %}
        {% set x_origin = x_default | float %}
        {% set y_origin = y_default | float %}
    {% endif %}
```
To this? 

`**{% set x_origin = ([x_origin, 230] | max) %}
        {% set y_origin = ([y_origin, 230] | max) %}**`

kyleisah, 2023-02-04:
I think there actually might be a way to make that happen. In the upper table you shared, there's a section that says `min` on 2 lines. I wonder if you change those to `max` if that would work?

Juste1911, 2023-02-04:
> I think there actually might be a way to make that happen. In the upper table you shared, there's a section that says `min` on 2 lines. I wonder if you change those to `max` if that would work?

I will try this sunday and let you know how it goes, thanks for the suggestion!

kyleisah, 2023-02-04:
I'm also curious if using both min and max or max and min or min min or max max would allow you to pick different corners specifically. Might be nice for edge cases

Juste1911, 2023-02-05:
> I'm also curious if using both min and max or max and min or min min or max max would allow you to pick different corners specifically. Might be nice for edge cases 

So, what I did was to set max in both lines where you suggested plus I set a minus (-) in front of both X Y variable distance to object...

Although I'm not fully convinced on how the pattern draws itself, moving away from the print so I will probably just set the purge to non variable (false) on my Enders and keep it variable (true) on my 2.4...

Anyway your work is excellent and thank you again for your time!

[image]

kyleisah, 2023-02-05:
Hey that's pretty neat! I'm impressed that the purge can be moved around that easily. I do have something else you can try. 

I have been working on a more generic purge line that is still adaptive, but is just a basic line. 

Would you be interested in testing it for me?

Juste1911, 2023-02-06:
> Hey that's pretty neat! I'm impressed that the purge can be moved around that easily. I do have something else you can try. 
> 
> I have been working on a more generic purge line that is still adaptive, but is just a basic line. 
> 
> Would you be interested in testing it for me?

Yes of course! Although I won't be printing (almost at all) this month because of university mid year exams/tests... If it's not a problem, regards!

kyleisah, 2023-02-19:
I have pushed a new purge to the repo that should help you with this issue, so I'm gonna go ahead and close this up 😄

closed 2023-02-19 by kyleisah