kyleisah/Klipper-Adaptive-Meshing-Purging #26: Feature request: Individual mesh per-object

whi-tw, 2023-02-06:
## Description

As the title says. It's probably something I'll noodle around with on my own, but I wonder: could we (optionally) create multiple meshes at start-time, one for each object on the build plate?

This would help get the nice tight meshes for each individual object. Specifically doing this when 'Complete individual objects' is on in SuperSlicer etc, would mean that each object has its own mesh.

## Implementation thoughts

1. Read the individual objects
2. for each object, create a named mesh (based on the name of the object)
3. Hijack `EXCLUDE_OBJECT_START` gcode, similar to:
```
[gcode_macro EXCLUDE_OBJECT_START]
rename_existing: EXCLUDE_OBJECT_START_BASE
gcode:
    {% set mesh_name = params.NAME %}
    BED_MESH_PROFILE LOAD={mesh_name}
    EXCLUDE_OBJECT_START {rawparams}
```

What are peoples' thoughts?

kyleisah, 2023-02-06:
If you can make that happen, I'd love to see it, but I don't think this is something I can dedicate time to

whi-tw, 2023-02-06:
> If you can make that happen, I'd love to see it, but I don't think this is something I can dedicate time to

For sure - makes complete sense. I just thought that here was the best place to record the idea. I'll have a go at it when I'm next trying to avoid something else!

whi-tw, 2023-02-08:
@kyleisah this may be it: https://github.com/whi-tw/klipper-adaptive-mesh/releases/tag/v0.0.1

kyleisah, 2023-02-12:
Hey, I've been pretty busy lately with work stuff, but I'm really impressed by this!

kyleisah, 2023-02-12:
So this is only really a function when you're doing sequential printing? Multiple objects on a single plate, printed one at a time? Also, have you worked out any toolhead crash prevention?

closed 2023-02-19 by kyleisah

Staubgeborener, 2023-04-15:
Well this topic was unfortunately closed, but is still relevant for me among others.
Since the original creator of this post is obviously no longer responding:

> So this is only really a function when you're doing sequential printing? Multiple objects on a single plate, printed one at a time?

Yes.

> Also, have you worked out any toolhead crash prevention?

The "toolhead crash prevention" is, at least for me, given by the slicer. Here a radius of the toolhead size was passed, so it crashes nowhere. Otherwise, this is not really an issue, as long as a bed mesh is created for all relevant areas (= all areas where objects are located) right from the start, before the very first object is printed. In this case there is no place where the toolhead could crash.

As you can see @kyleisah, this is still a problem and an important feature. This issue should be reopened.

whi-tw, 2023-04-15:
I'm not specifically "no longer responding", I just didn't see the notifications from this thread.

I'm not certain if the slicer's toolhead crash prevention would necessarily save you here - as the mesh area is min/max x by min/max y. The crash prevention works on the actual bounds of the object, and the mesh area will extend outside of this.

I've been using my fork to create the meshes as part of my `PRINT_START` macro - as @Staubgeborener states, this is the true crash prevention: no moves are forced when there's any plastic on the bed.

kyleisah, 2023-04-15:
@Staubgeborener I will have to finish current dev before I can look into this again. This seems a little niche, but I can see how it can be a benefit for those who want it.