kyleisah/Klipper-Adaptive-Meshing-Purging #27: Not working on a Delta

ManCubFPV, 2023-02-09:
At first is said it was getting an error with probe_count, i changed everything that said probe_count to round_probe_count, error isn't showing, but now I get !! Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: object of type 'int' has no len()"

kyleisah, 2023-02-09:
Hey there, so, I don't have a delta printer, and I know very little about how round meshing works, so I'm sorry to say you might be on your own for this one. But, if you share your klipper log file, I might be able to see if there's something easy

ManCubFPV, 2023-02-09:
The only difference that I can see for bed meshing on a delta is
probe_count is "round_probe_count" on a delta and it only uses one number.
I use "5" so it probes 5 points in the middle and gradually less, working
its way outwards towards the edge of the plate.

round bed, round_probe_count = 5, bed_radius = r:
                 x (0, r) end
               /
             x---x---x
                       \
 (-r, 0) x---x---x---x---x (r, 0)
           \
             x---x---x
                   /
                 x  (0, -r) start

Also there is no min or max, just origin which is always 0,0 (center
of the plate) so it can go into negative coordinates.

Lastly there's radius, self-explanatory.

I'm willing to be a guinea pig if you have ideas, or maybe just point
me in the right direction of proper coding. I'm not a coder, but I can
try to learn for sure.

ManCubFPV, 2023-02-09:
> Hey there, so, I don't have a delta printer, and I know very little about how round meshing works, so I'm sorry to say you might be on your own for this one. But, if you share your klipper log file, I might be able to see if there's something easy

The only difference that I can see for bed meshing on a delta is probe_count is "round_probe_count" on a delta and it only uses one number. I use "5" so it probes 5 points in the middle and gradually less, working its way outwards towards the edge of the plate.

round bed, round_probe_count = 5, bed_radius = r:
                 x (0, r) end
               /
             x---x---x
                       \
 (-r, 0) x---x---x---x---x (r, 0)
           \
             x---x---x
                   /
                 x  (0, -r) start

Also there is no min or max, just origin which is always 0,0 (center of the plate) so it can go into negative coordinates.

Lastly there's radius, self-explanatory.

I'm willing to be a guinea pig if you have ideas, or maybe just point me in the right direction of proper coding. I'm not a coder, but I can try to learn for sure.

kyleisah, 2023-02-12:
Hey there, I'm having another look at this right now because I have some free time. I took a look at how klipper handles round meshes and it look like this may be possible. It works by setting `MESH_ORIGIN` and `MESH_RADIUS`. So the origin would be the center of the print area being used, and the radius would be the distance from the center of the print area to the edge of the objects.

Are you by any chance on Discord? I'd love if we can chat and try some stuff out.

ManCubFPV, 2023-02-12:
Bigfoot Cowboy#8006 is my discord username

kyleisah, 2023-02-12:
Hey, sorry, had to delete your comment to prevent you getting doxxed haha. I'll hit you up on discord.

kyleisah, 2023-02-12:
@ManCubFPV Not sure why I can't find you, send me a friend request or DM directly, not sure how that works exactly, but my Discord ID is Kyleisah#2433

kyleisah, 2023-02-27:
```jinja
[gcode_macro _KAMP_Get_Delta_Params]
description: HELPER - Gets the origin and radius values for adaptive meshing with delta printers
gcode:

    # Get all defined object points.
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}
 
    # Get XY min/max area from object bounds.
    {% set x_min = all_points | map(attribute=0) | min %}
    {% set y_min = all_points | map(attribute=1) | min %}
    {% set x_max = all_points | map(attribute=0) | max %}
    {% set y_max = all_points | map(attribute=1) | max %}

    # Convert bounding box to center point (round mesh origin).
    {% set x_origin = (x_max - x_min) / 2 %}
    {% set y_origin = (y_max - y_min) / 2 %}
    
    # Get XY origin point and radius from [bed_mesh] configuration.
    {% set default_delta_origin_x = printer.configfile.settings.bed_mesh.mesh_origin[0] %}
    {% set default_delta_origin_y = printer.configfile.settings.bed_mesh.mesh_origin[1] %}
    {% set default_delta_radius = printer.configfile.settings.bed_mesh.mesh_radius %}
    
    # Calculate radius values to object points from mesh_origin and pick the largest one.
    {% for point in all_points %}
        {% set distance = ((point[0]-x_origin) ** 2 + (point[1]-y_origin) ** 2) ** 0.5 %}
        {% set max_distance = [max_distance|default(0), distance] | max %} 
    {% endfor %}

    # If calculated radius is smaller than default mesh_radius, use calculated radius. If not, use default.
    {% if max_distance < default_delta_radius %}
        {% set delta_radius = max_distance %}
    {% else %}
        {% set delta_radius = default_delta_radius %}
        {% set x_origin = default_delta_origin_x %}
        {% set y_origin = default_delta_origin_y %}
    {% endif %}

    # Send calculated values to Adaptive Mesh macro.
    # SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_origin VALUE={[x_origin, y_origin]}
    # SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_radius VALUE={delta_radius}

    { action_respond_info("Origin found at [{!r}], radius calculated is {}".format(
        (x_origin, y_origin),
        delta_radius,
    )) }```

Add this macro to your printer.cfg anywhere it'll get loaded. Then, restart and use this command in the console:
`EXCLUDE_OBJECT_DEFINE NAME=calibration_pyramid CENTER=50,50 POLYGON=[[40,40],[50,60],[60,40]]`

Finally, run the command `_KAMP_GET_DELTA_PARAMS` and it should spit out an origin and a radius

kyleisah, 2023-02-27:
@julianschill Just tagging you so you can follow along with this. We ran the macro to just output an origin and a radius in the console for testing, but it looks like the `max_distance` param is having an issue. An exception gets thrown which says it's not defined, which kind of makes sense because max_distance is getting defined from max_distance.

Here's the error:

[image]

Any ideas?

julianschill, 2023-02-27:
I will look into it. We probably just have to initialize the variable.

dangerzau, 2023-03-04:
So I defined the variable in the macro just to test (Very little idea what I'm doing here, just want this to work) 
And un-commented the two set commands as below
```
# SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_origin VALUE={[x_origin, y_origin]}
# SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_radius VALUE={delta_radius}
```

Now get the following output using the same test commands mentioned 
```
EXCLUDE_OBJECT_DEFINE NAME=calibration_pyramid CENTER=50,50 POLYGON=[[40,40],[50,60],[60,40]]
_KAMP_GET_DELTA_PARAMS
```
6:20 pm
Malformed command 'SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_origin VALUE=[10.0, 10.0]'
6:20 pm
Malformed command 'SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_origin VALUE=[10.0, 10.0]'
6:20 pm
Origin found at [(10.0, 10.0)], radius calculated is 0 <-- Clearly I've not helped - but at least the macro progresses I guess

```
[gcode_macro _KAMP_Get_Delta_Params]
description: HELPER - Gets the origin and radius values for adaptive meshing with delta printers
variable_max_distance: 0
gcode:

    # Get all defined object points.
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}
 
    # Get XY min/max area from object bounds.
    {% set x_min = all_points | map(attribute=0) | min %}
    {% set y_min = all_points | map(attribute=1) | min %}
    {% set x_max = all_points | map(attribute=0) | max %}
    {% set y_max = all_points | map(attribute=1) | max %}

    # Convert bounding box to center point (round mesh origin).
    {% set x_origin = (x_max - x_min) / 2 %}
    {% set y_origin = (y_max - y_min) / 2 %}
    
    # Get XY origin point and radius from [bed_mesh] configuration.
    {% set default_delta_origin_x = printer.configfile.settings.bed_mesh.mesh_origin[0] %}
    {% set default_delta_origin_y = printer.configfile.settings.bed_mesh.mesh_origin[1] %}
    {% set default_delta_radius = printer.configfile.settings.bed_mesh.mesh_radius %}
    
    # Calculate radius values to object points from mesh_origin and pick the largest one.
    {% for point in all_points %}
        {% set distance = ((point[0]-x_origin) ** 2 + (point[1]-y_origin) ** 2) ** 0.5 %}
        {% set max_distance = [max_distance | default(0), distance] | max %} 
    {% endfor %}

    # If calculated radius is smaller than default mesh_radius, use calculated radius. If not, use default.
    {% if max_distance < default_delta_radius %}
        {% set delta_radius = max_distance %}
    {% else %}
        {% set delta_radius = default_delta_radius %}
        {% set x_origin = default_delta_origin_x %}
        {% set y_origin = default_delta_origin_y %}
    {% endif %}

    # Send calculated values to Adaptive Mesh macro.
     SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_origin VALUE={[x_origin, y_origin]}
     SET_GCODE_VARIABLE MACRO=Adaptive_Mesh VARIABLE=delta_radius VALUE={delta_radius}

    { action_respond_info("Origin found at [{!r}], radius calculated is {}".format(
        (x_origin, y_origin),
        delta_radius,
    )) }

```

kyleisah, 2023-03-04:
Yeah, those commands were commented out because they're not going to work right now. I just wanted to get an output that shows an origin and radius was calculated

dangerzau, 2023-03-05:
If radius is showing as 0 - Does that mean it did in fact not calculate one based on the initial line
`EXCLUDE_OBJECT_DEFINE NAME=calibration_pyramid CENTER=50,50 POLYGON=[[40,40],[50,60],[60,40]]
`
For reference, my "default" mesh_radius is 73

hmmmm... wonder if I should/can (Again only new to this) initialise that variable to the configfile mesh_radius rather than 0 - at least then a fallback is always going to be the default if things go wrong.

kyleisah, 2023-03-05:
> If radius is showing as 0 - Does that mean it did in fact not calculate one based on the initial line `EXCLUDE_OBJECT_DEFINE NAME=calibration_pyramid CENTER=50,50 POLYGON=[[40,40],[50,60],[60,40]] ` [...]

Give me some time and I will cook up a .zip file for you that will contain everything necessary to try it, I've been busy with other things KAMP related but I will do my best!

kyleisah, 2023-03-05:
Unless you think you can try it out. Maybe it *did* in fact adjust your radius to zero?

snimax, 2023-04-17:
Is there any new updates on this issue or is the latest version the one you sent on Feb27? I'm interested in using it for my delta😄

iqubik, 2023-04-17:
purge work correctly on delta flsun q5

snimax, 2023-04-18:
> purge work correctly on delta flsun q5

Nice thats good to know 👍 

> > Is there any new updates on this issue or is the latest version the one you sent on Feb27? I'm interested in using it for my delta😄
> 
> Hey there! Yes! There has been some progress on this. I currently have a semi-working macro for adapting a delta printer's mesh.
> 
> It's very basic right now, but I have the bones of it built. I just have to cook the rest of KAMP's goodness into it.

Oh that's sounds promising 😀 I could help out with testing if needed

kyleisah, 2023-04-20:
@iqubik what error are you getting? Bear in mind, KAMP is not ready for delta printers *yet*

tcj-lev, 2023-04-20:
I just wanted to try out the dev branch
So I've uploaded the macros from dev Configuration onto my heavy modded Anycubic Predator.
But it does not work yet
At first I had to comment out the three include lines of KAMP_Settings.cfg, because this folder and files don't exist
after that I had to comment out the gcode of the same file, because of:
`Error loading template 'gcode_macro _KAMP_Settings:gcode': TemplateSyntaxError: unexpected char u'$' at 130
`
After that I got an `Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: No aggregated item, sequence was empty.`
Delta_Mesh.cfg is the only file containing a [gcode_macro BED_MESH_CALIBRATE]

What did I do wrong?

kyleisah, 2023-04-20:
> I just wanted to try out the dev branch So I've uploaded the macros from dev Configuration onto my heavy modded Anycubic Predator. But it does not work yet At first I had to comment out the three include lines of KAMP_Settings.cfg, because this folder and files don't exist after that I had to comment out the gcode of the same file, because of: `Error loading template 'gcode_macro _KAMP_Settings:gcode': TemplateSyntaxError: unexpected char u'$' at 130 ` After that I started a test print, but got an `Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: No aggregated item, sequence was empty.` when the BED_MESH_CALIBRATE was called by the print_start macro Delta_Mesh.cfg is the only file containing a [gcode_macro BED_MESH_CALIBRATE]
> 
> What did I do wrong?

I am not debugging dev macros on other machines at this time. They are not deemed ready yet.

tcj-lev, 2023-04-20:
> I am not debugging dev macros on other machines at this time. They are not deemed ready yet.

ok
is there any way that i could help with testing/debugging?

tcj-lev, 2023-04-22:
Hi,
I've had a closer look into your code regarding Delta-Meshing.
This might not work in special cases.
When the print/mesh-area is off the centre (0,0) and you are calculating a new mesh_origin and also calcuting a new mesh_radius you might leave the print-bed.
To make it easy visible:
take a longer object like a ruler and place it on any round surface (like a delta printbed) so that the edges are roughly near the circumference. 
Now turn the objet by 90° around the center of the object
do you see what i mean?

closed 2023-07-06 by kyleisah