kyleisah/Klipper-Adaptive-Meshing-Purging #28: Question: S3D MESH_CONFIG functionality

Zarrsito, 2023-02-12:
Hello,

i love to use this plugin =) and i love to use Simplify3D... well but simplify3d isnt good for labeling objects in gcode, however
they implemented a new function (MESH_CONFIG X0=[build_min_x] Y0=[build_min_y] X1=[build_max_x] Y1=[build_max_y]) will this Script be compatible to this function??? or maybe u can addapt this function into your script making simplify3d supported finally ! :)

kyleisah, 2023-02-12:
I imagine something *might* be possible here. S3D has a scripting engine and it seems you could use those values you've mentioned to define a mesh area. Unfortunately, I don't use Simplify3D so I won't be able to test it myself. I'll see if there's something that can be done. 

I think it's possible to make it work, but you wouldn't be using [exclude_object] processing to make it happen. You'd be passing values from the slicers output to the macro's input kind of how adaptive meshing was traditionally done before.

Zarrsito, 2023-02-12:
Hey, i already looked a little bit further into this, but its not possible for me to adapt things xD
there is another plugin called PAM for RatOS, i already digged into that, but im an electrician not an programmer :(
https://github.com/HelgeKeck/pam/blob/main/klippy_extra/pam.py 
there is all the magic i guess... 

U wont need simplify3d for testing, the output from simplify3d is simple like this

MESH_CONFIG X0=94.11 Y0=39.06 X1=140.94 Y1=195.94

i think u just need to hook onto this :)

kyleisah, 2023-02-12:
I really fell down a rabbit hole haha, I also found that PAM script in all my searching, so that's funny. I believe I have something that will work!

```ini
[gcode_macro MESH_CONFIG]
gcode:
    {% set X0 = params.X0 | float %}
    {% set X1 = params.X1 | float%}
    {% set Y0 = params.Y0 | float%}
    {% set Y1 = params.Y1 | float%}
    {% set center_x = ((X0 + X1) / 2) | float %}
    {% set center_y = ((Y0 + Y1) / 2) | float %}

    EXCLUDE_OBJECT_DEFINE NAME=s3d_build_volume CENTER={center_x},{center_y} POLYGON=[[{X0},{Y0}],[{X0},{Y1}],[{X1},{Y1}],[{X1},{Y0}]] ```

Add that macro to your config, and add this to your slicer's start gcode *before* your `PRINT_START` macro is called:

```ini
MESH_CONFIG X0=[build_min_x] Y0=[build_min_y] X1=[build_max_x] Y1=[build_max_y] ```

I believe that will make it work. Please test this for me. What the above macro does is take the output from your slicer for the used build area and converts that area to 1 object to be used for `[exclude_object]` which will make S3D work "good enough" for KAMP. :)

Zarrsito, 2023-02-12:
hahahaha OMG it works ! :D thats exactly what i was thinking of how the code could possibly look like xD YAY you have S3D support :D where is ur Tip Bin xD

kyleisah, 2023-02-12:
That's awesome to hear dude! I'm glad it works. I'll go ahead and close this up for now and add it to the upcoming overhaul branch.

closed 2023-02-12 by kyleisah