kyleisah/Klipper-Adaptive-Meshing-Purging #29: Feature Request: Mesh Margin

spiro2k6, 2023-02-15:
A small issue I've come across is that SuperSlicer does not count the Brim when labeling objects.  When printing some things in ABS i use a large brim (12mm) to help keep the first layer down on my non enclosed printer.  2 things happen with this, first the adaptive purge ends up in the path of my skirt.  This isn't a big deal as i adjusted the purge distance to be farther from the print.  The bigger issue is that I will get my z offset set while the large brim is printing, but it is outside of the adaptive meshing area, and after it gets to the actual print and the mesh takes over my z offset ends up way too low for the first layer.  So i end up with good brim and a oversquished first layer on the print, or a bad brim that is too high and a good first layer for the print area  (the printer with this issue has a very warped 400x400mm bed)

All this is to ask if a Margin variable can be added to the macro so that you can specify in MM an area around the calculated mesh to be added to the probing area.  (for example setting it to 10 would make the calculated mesh and probing points 10mm larger in all directions.)

I was going to try to hardcode in a value to the existing macro but wasnt sure how to do this and also have it still verify the new probe points are within printable area/mesh min and max.

I'm sorry if this is too longwinded, I just feel like this could be a useful feature for many people that may be very simple to implement for someone familiar with the codebase.

kyleisah, 2023-02-15:
Hm, this is an interesting idea, I don't think this would be terribly hard to do. Let me see what I can find

kyleisah, 2023-02-15:
Alright, so I have one area of concern. One thing I feel is mandatory with KAMP is "If it slices, the mesh will succeed."

When we add margins, there's a chance the meshable area can be exceeded and klipper will throw errors. There's probably more than can be worked into it so there's safeguards, but this is a very specific usecase.

That being said, I have something you are welcome to try.

```ini
# # # Klipper Adaptive Meshing # # #

# Heads up! If you have any other BED_MESH_CALIBRATE macros defined elsewhere in your config, you will need to comment out / remove them for this to work. (Klicky/Euclid Probe)
# You will also need to be sure that [exclude_object] is defined in printer.cfg, and your slicer is labeling objects.
# This macro will parse information from objects in your gcode to define a min and max mesh area to probe, creating an adaptive mesh!
# This macro will not increase probe_count values in your [bed_mesh] config. If you want richer meshes, be sure to increase probe_count. We recommend at least 5,5.

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: _BED_MESH_CALIBRATE

### This section allows control of status LEDs your printer may have.

variable_led_enable: False              # Enables/disables the use of status LEDs in this macro.
variable_status_macro: 'status_meshing' # If you have status LEDs in your printer (StealthBurner), you can use the macro that changes their status here.

### This section configures mesh point fuzzing, which allows probe points to be varied slightly if printing multiples of the same G-code file.

variable_fuzz_enable: False             # Enables/disables the use of mesh point fuzzing to slightly randomize probing points to spread out wear on a build surface, default is False.
variable_fuzz_min: 0                    # If enabled, the minimum amount in mm a probe point can be randomized, default is 0.
variable_fuzz_max: 4                    # If enabled, the maximum amount in mm a probe point can be randomized, default is 4.

variable_margin_enable: False           # Enables/disables adding a margin to the meshed area to pad a mesh out for specific needs, default is False.
variable_margin_size: 5                 # Size in millimeters to expand the mesh outwards from the print area in both directions.

### This section is for those using a dockable probe that is stored outside of the print area. ###

variable_probe_dock_enable: False       # Enables/disables the use of a dockable probe that is stored outside of the print area, default is False.
variable_attach_macro: 'Attach_Probe'   # Here is where you define the macro that ATTACHES the probe to the printhead. E.g. 'Attach_Probe'
variable_detach_macro: 'Dock_Probe'     # Here is where you define the macro that DETACHES the probe from the printhead. E.g. 'Dock_Probe'

gcode:
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}
    {% set bed_mesh_min = printer.configfile.settings.bed_mesh.mesh_min %}
    {% set bed_mesh_max = printer.configfile.settings.bed_mesh.mesh_max %}
    {% set probe_count = printer.configfile.settings.bed_mesh.probe_count %}
    {% set probe_count = probe_count if probe_count|length > 1 else probe_count * 2  %}
    {% set max_probe_point_distance_x = ( bed_mesh_max[0] - bed_mesh_min[0] ) / (probe_count[0] - 1)  %}
    {% set max_probe_point_distance_y = ( bed_mesh_max[1] - bed_mesh_min[1] ) / (probe_count[1] - 1)  %}
    {% set x_min = all_points | map(attribute=0) | min | default(bed_mesh_min[0]) %}
    {% set y_min = all_points | map(attribute=1) | min | default(bed_mesh_min[1]) %}
    {% set x_max = all_points | map(attribute=0) | max | default(bed_mesh_max[0]) %}
    {% set y_max = all_points | map(attribute=1) | max | default(bed_mesh_max[1]) %}
    
    { action_respond_info("{} object points, clamping to bed mesh [{!r} {!r}]".format(
        all_points | count,
        bed_mesh_min,
        bed_mesh_max,
    )) }

    {% if fuzz_enable == True %}
        {% set fuzz_range = range(fuzz_min * 100 | int, fuzz_max * 100 | int + 1) %}
        {% set x_min = (bed_mesh_min[0] + fuzz_max, x_min) | max - (fuzz_range | random / 100.0) %}
        {% set y_min = (bed_mesh_min[1] + fuzz_max, y_min) | max - (fuzz_range | random / 100.0) %}
        {% set x_max = (bed_mesh_max[0] - fuzz_max, x_max) | min + (fuzz_range | random / 100.0) %}
        {% set y_max = (bed_mesh_max[1] - fuzz_max, y_max) | min + (fuzz_range | random / 100.0) %}
    {% else %}
        {% set x_min = [ bed_mesh_min[0], x_min ] | max %}
        {% set y_min = [ bed_mesh_min[1], y_min ] | max %}
        {% set x_max = [ bed_mesh_max[0], x_max ] | min %}
        {% set y_max = [ bed_mesh_max[1], y_max ] | min %}
    {% endif %}

    {% if margin_enable == True %}
[... 25 lines cut ...]
    { action_respond_info( "Points: x: {}, y: {}".format(points_x, points_y) ) }

    {% if printer.configfile.settings.bed_mesh.relative_reference_index is defined %}
        {% set ref_index = (points_x * points_y / 2) | int %}
        { action_respond_info( "Reference index: {}".format(ref_index) ) }  
    {% else %}
        {% set ref_index = -1 %}
    {% endif %}

    {% if probe_dock_enable == True %}
        {attach_macro}              # Attach/deploy a probe if the probe is stored somewhere outside of the print area
    {% endif %}

    {% if led_enable == True %}
        {status_macro}              # Set status LEDs
    {% endif %}

    _BED_MESH_CALIBRATE mesh_min={x_min},{y_min} mesh_max={x_max},{y_max} ALGORITHM={algorithm} PROBE_COUNT={points_x},{points_y} RELATIVE_REFERENCE_INDEX={ref_index}

    {% if probe_dock_enable == True %}
        {detach_macro}              # Detach/stow a probe if the probe is stored somewhere outside of the print area
    {% endif %}
```

Hopefully this helps!

spiro2k6, 2023-02-15:
Would changing this to below ensure that if the margin would put it outside the defined mesh min or max that it would just use that min or max instead for meshing on each side.
`    {% if margin_enable == True %}
        {% set x_min = [ bed_mesh_min[0], x_min - margin_size ] | max %}
        {% set y_min = [ bed_mesh_min[1], y_min - margin_size ] | max %}
        {% set x_max = [ bed_mesh_max[0], x_max + margin_size ] | min %}
        {% set y_max = [ bed_mesh_max[1], y_max + margin_size ] | min %}
    {% endif %}`

kyleisah, 2023-02-15:
lmao that's literally what I just did :joy:

kyleisah, 2023-02-15:
Try both and let me know what you come up with!

spiro2k6, 2023-02-15:
Lol, I was looking over how you did it for the main part and it clicked.  This is perfect though, it should get me exactly what I was looking for.  Thank you so much, I appreciate you responding so quickly and helping out.  I will try them tomorrow and let you know.  I've got to be up in 6 hours so i'm gonna call it.  I found you on the Voron discord so i can reach out there if it helps keep this page more tidy.

kyleisah, 2023-02-15:
Feel free to DM me and we will figure it out! 😄

closed 2023-02-17 by kyleisah