kyleisah/Klipper-Adaptive-Meshing-Purging #35: Error evaluating 'gcode_macro BED_MESH_CALIBRATE

hmouridsen, 2023-02-21:
Hi

I'm trying to make KAMP working - however I get the following error?

I'm on TAP incase that matters..

From print start:
STATUS_LEVELING
    G32; home and QGL
    SETUP_KAMP_MESHING DISPLAY_PARAMETERS=1 LED_ENABLE=1 FUZZ_ENABLE=1
    BED_MESH_CLEAR                 ; remove any existing bed mesh (EDIT)
    STATUS_MESHING
    BED_MESH_CALIBRATE             ; generate bed mesh (EDIT)
    
    
output:
21.14
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: integer argument expected, got float
21.14
Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: integer argument expected, got float
21.14
4 object points, clamping to bed mesh [(40.0, 40.0) (310.0, 310.0)]
21.14
detach_macro: 'Dock_Probe'
21.14
attach_macro: 'Attach_Probe'
21.14
probe_dock_enable: 0
21.14
fuzz_max : 4.000000
21.14
fuzz_min : 0.000000
21.14
fuzz_enable : 1
21.14
status_macro: 'status_meshing'
21.14
led_enable : 1

kyleisah, 2023-02-21:
Hmmm, looks like it's something to do with klipper expecting an `integer` value and getting a `float` value. Let me see what I can find.

kyleisah, 2023-02-21:
I believe this was fixed with PR #34! Let me know if it works for you now.

hmouridsen, 2023-02-22:
> I believe this was fixed with PR #34! Let me know if it works for you now.

Thanks a million- the update did fix my problem.

Thanks for sharing your work 👍🏻

closed 2023-02-22 by hmouridsen