kyleisah/Klipper-Adaptive-Meshing-Purging #36: Incorrect purge move speed

310weber, 2023-03-01:
In Line_Purge.cfg -Line 51 
```  {% set purge_move_speed = (cross_section * flow_rate) * 60 | float %}```

This yeilds a number with units mm^5/sec (mm^2 * mm^3/sec)

I believe it should be changed to the following:
```  {% set purge_move_speed = (flow rate / cross_section) * 60 | float %}```

kyleisah, 2023-03-01:
That's what I get for working on new purges late at night I suppose. 😅 I'm actually working on revamping all of the macros a bit to improve the experience of people using Moonraker's Update Manager. This will also improve the experience of those who go the more traditional route though as well. 

I've made some helper macros that will just do the calculations and then push them all purge macros, and whatever purge macro is called will be able to verbose some good information. This also led me down the path of needing to rework how flow_rate is calculated.

I knew something was funky about the way it was calculated in `LINE_PURGE` but it still worked on my machines just fine.

If you'd please check my work here, this is how purge_move_speed (flow) will be calculated in an upcoming release. Remember, these are not finalized, but is just a way to get what was in my mind into vscode:

The variables for purging here:
```yaml
# The following variables are for adjusting PURGE settings for KAMP.
variable_adaptive_purging_enable: False     # Set to True to enable adaptive purging with KAMP.
variable_tip_distance: 0                    # Distance between tip of filament and nozzle before purge. Useful for reliable and consistent purging.
variable_y_distance_to_object:  5           # Distance between purge line and print area.
variable_flow_rate: 12                      # Desired flow rate for purge. Standard flow hotends are ~12, high flow are ~24.
variable_purge_amount: 25                   # Amount in millimeters of filament to purge.
```

The helper macro here:
```yaml
[gcode_macro _KAMP_Purge_Calculate]
description: HELPER - Gets any params KAMP requires for purging and calculates purge variables
gcode:

    # Get nozzle diameter from printer.cfg.
    {% set nozzle_dia = printer.configfile.config.extruder.nozzle_diameter | float %}

    # Set purge Z height to be 75% of nozzle diameter.
    {% set z_height = nozzle_dia * 0.75 %}
    {% set purge_speed = (flow_rate / nozzle_dia / z_height) * 60 | float %}
```

kyleisah, 2023-03-01:
Hmm, now that I'm looking at this again, I think my math is wrong.

kyleisah, 2023-03-01:
No, no, I believe that's correct.

If `flow_rate` is `12` and the line_width(`nozzle_dia`) is 0.4, and the `z_height` is 0.3, that would mean the fastest you could purge is at a speed of 100mm/s. 100mm/s * 60, 6000mm/m would be the feedrate.

310weber, 2023-03-01:
That looks correct to me but it makes the assumption that the purge line is only as wide as the nozzle diameter.  

As I looked through the current line purge macro more I came to the conclusion it is over constrained.  The user can specify purge amount, nozzle height, and line length, and the line width is assumed to be equal to nozzle diameter. The reality is that if you increase purge amount without increasing line length or nozzle height, the line will ultimately just squish out wider. The problem is that the speed calculation uses the assumed line width (cross section) so you'll end up inadvertently increasing flow rate because the speed calculation doesn't know how wide the line will actually be. 

I was thinking along the same lines as your helper macros for common extrude calculations. My plan was to have two:
- One would allow you to specify purge amount (along with flow rate, nozzle height, and line width), and it would output line length and speed. It would throw an error if line length exceeds some maximum, e.g. the bed length - start position. 
- The second would allow you to specify purge line length (along with flow rate, nozzle height and line width), and it would output the purge amount and speed. It would throw an error if purge amount exceeds the klipper max of 50 mm.

kyleisah, 2023-03-01:
One thing I'm considering is just defining amount to purge, and flow rate, and the macro will take the x_min and x_max of the print area, move up y_distance_to_object, and then work out a decent purge line from those constraints. It's really hard to get this to work for any possible combination *and* keep defined variables to a minimum. 

Maybe something like `flowrate / purge_amount * calculated_length_from_print_area`?

310weber, 2023-03-02:
That sounds like a good aproach for this application.  People are likely to be more interested in specifying the amount of filament to be purged rather than the length of the line.  Just take the flow rate and amount as input and let the macro calculate reasonable values for the rest.

I'll link to whatever I come up with in the next couple days and am happy to review your solution when it's ready as well.

kyleisah, 2023-03-02:
I'd love another pair of eyes on this if you want to give it a shot!

Any purge macro/method you come up with, these are the variables that are being standardized for all purges:

```yaml
variable_tip_distance: 0
variable_y_distance_to_object: 0
variable_flow_rate: 0
variable_purge_amount: 0
```
Looking forward to seeing what you come up with!

310weber, 2023-03-12:
Sorry it took more that a couple days but here's what I came up with.  
https://github.com/310weber/klipper_macros/blob/main/purge_underline.cfg

My method is to essentially "underline" the active print area - adjusting the thickness of the purge line so the correct amount of filament is extruded across the entire bottom edge of the print.  So, if you're printing something small it will be a short, thick line.  If you're printing something larger, the purge line will be thinner and longer to match.  There are checks to safegard against lines that are too thin/thick or extending off the edge of the print surface.  Let me know what you think.

mpetito, 2023-03-22:
I think I've run into this issue with LINE_PURGE macro in main.  Klipper is complaining about exceeding maximum extrusion with a relatively small print area.  You can see I tried cutting the flow rate in half (from 12 to 6) but oddly enough that did not seem to change the error.

> Move exceeds maximum extrusion (1.924mm^2 vs 1.440mm^2)
See the 'max_extrude_cross_section' config option for details
distance_to_object_y : 10.000000
y_default : 10.000000
x_default : 10.000000
flow_rate : 6.000000
line_length : 50.000000
purge_amount : 40.000000
z_height : 0.400000
adaptive_enable : 1

ogabriel, 2023-03-26:
I have also run in to the issue with maximum extrude, but with the VORON_PURGE and a 0.4mm nozzle

> Move exceeds maximum extrusion (4.165mm^2 vs 0.640mm^2)
See the 'max_extrude_cross_section' config option for details

Kaan88, 2023-03-31:
> Sorry it took more that a couple days but here's what I came up with. https://github.com/310weber/klipper_macros/blob/main/purge_underline.cfg
>
> My method is to essentially "underline" the active [...]

Any reason why you took it down? I wanted to give it a try.

310weber, 2023-03-31:
> Any reason why you took it down? I wanted to give it a try.

Just a mistake - I set the repo to private by mistake.  Should be good now.

mpetito, 2023-03-31:
I took a somewhat different approach here: https://github.com/mpetito/printer-scripts/blob/main/macros/line_purge.cfg

It determines where to draw a series of connected lines (above, below, left or right of the print area based on maximizing available area), and calculates the number of lines to draw based on a target flow rate and purge amount.

This is my first klipper macro and deep dive into gcode but it has been working well and the flow calculations seem to work correctly.

FiyaFly, 2023-04-08:
> I have also run in to the issue with maximum extrude, but with the VORON_PURGE, a 0.4mm nozzle and a small print
> 
> > Move exceeds maximum extrusion (4.165mm^2 vs 0.640mm^2)
> > See the 'max_extrude_cross_section' config option for details
> 
> Fixed by adding this to `[extruder]`:
> 
> > max_extrude_cross_section: 5

I'm planning on digging deeper into this in the next couple days, but I did want to add in a note for these macros that I've ran across, since it seems there are a few people taking a look at this. There seems to be two general issues to handle here, and I'm not sure if we want to keep them both under this one, or create another one as well:

- Purge move speed
- max_extrude_cross_section

Noticed this in the Line_Purge.cfg as well, tried adding it, and it worked. _However_, on transitioning to the second layer of my print, I had this error again as well, which told me something was sticking around from this purge macro. Easy fix to at least get past this part. At the beginning of the GCode for the macro, add `SAVE_GCODE_STATE NAME=Line_Purge_state`, and at the end, `RESTORE_GCODE_STATE NAME=Line_Purge_state`. It might not resolve the overall issue, but at least keeps it contained.

Just adding it here so anyone working on tweaks on this can add this, without me just adding it to one section and it getting missed by someone else.

I'm going to take a deeper look at the feed rate and purge move speed calculations in the next couple days as well.

FiyaFly, 2023-04-08:
> I'm planning on digging deeper into this in the next couple days, but I did want to add in a note for these macros that I've ran across, since it seems there are a few people taking a look at this. [...]

Couldn't help it. Bad at letting things go. lol. Looks like just a bad translation of the formula for flow rate.

In the config, it shows `v = a*f` (multiplied by 60 for setting the feedrate as per min, which is right).  
The original formula is `f=a*v`, so I think we need to convert it to `v=f/a` so:

`{% set purge_move_speed = (cross_section * flow_rate) * 60 | float %}`  
turns into  
`{% set purge_move_speed = (flow_rate / cross_section) * 60 | float %}`

I can throw in the changes and do a pull request tomorrow.

kyleisah, 2023-04-08:
I appreciate all of you helping me out on this! KAMP is undergoing a fairly large overhaul in dev right now. I'm almost finished with getting the verbosity just right, and reworking the macros a bit to make a bit more sense to anyone else looking at it (including me!). Once I get the mesh stuff out of the way, I will start working on the purge stuff. That being said, I would love to see some PRs of fixes you guys come up with, but I cannot promise they will be merged in their current state until the dev stuff is done. Thanks again, and keep on KAMPing! 😄

FiyaFly, 2023-04-08:
> I appreciate all of you helping me out on this! KAMP is undergoing a fairly large overhaul in dev right now. I'm almost finished with getting the verbosity just right, and reworking the macros a bit [...]

I noticed that as I was looking around. That's exactly why I opted to post here first 😄 Let me know the best ways I can help and I'm happy to give what I can!

yebo29, 2023-04-19:
> I have also run in to the issue with maximum extrude, but with the VORON_PURGE, a 0.4mm nozzle and a small print
>
>
> Fixed by adding this to `[extruder]`:

Adding that to my extruder config did the trick for me. Thank you!

kyleisah, 2023-04-20:
@yebo29 The next release of KAMP will warn you if you don't have that configured correctly, and skip the purge. 😄 

[image]

kyleisah, 2023-04-20:
@FiyaFly Hey, I really wanted to thank you for what your shared in terms of calculating the purge's movement speed. Using the max_cross_section value was pretty damn clever, and it seems to work nicely. That was a major pain for me to get it just right, and I think it's way better than it was before.

Here's the **DEVELOPMENT** macro if you're interested, for Line_Purge. It wont work without having a _KAMP_Settings macro (this is to greatly improve moonraker/setup-macro woes) but it's a nice sneak peak. 😉 

```jinja
[gcode_macro LINE_PURGE]
description: A purge macro that adapts to be near your actual printed objects
gcode:
    # Get relevant printer params
    {% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}
    {% set cross_section = printer.configfile.settings.extruder.max_extrude_cross_section | float %}
    
    # Use firmware retraction if it is defined
    {% if printer.firmware_retraction is defined %}
        {% set RETRACT = G10 | string %}
        {% set UNRETRACT = G11 | string %}
    {% else %}
        {% set RETRACT = 'G1 E-.5 F2100' | string %}
        {% set UNRETRACT = 'G1 E.5 F2100' | string %}
    {% endif %}

    # Get purge settings from _Kamp_Settings
    {% set verbose_enable = printer["gcode_macro _KAMP_Settings"].verbose_enable | abs %}
    {% set purge_height = printer["gcode_macro _KAMP_Settings"].purge_height | float %}
    {% set tip_distance = printer["gcode_macro _KAMP_Settings"].tip_distance | float %}
    {% set purge_margin = printer["gcode_macro _KAMP_Settings"].purge_margin | float %}
    {% set purge_amount = printer["gcode_macro _KAMP_Settings"].purge_amount | float %}
    {% set flow_rate = printer["gcode_macro _KAMP_Settings"].flow_rate | float %}

    # Calculate purge origins and centers from objects
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}    # Get all object points
    {% set purge_x_min = (all_points | map(attribute=0) | min | default(0)) %}                          # Object x min
    {% set purge_x_max = (all_points | map(attribute=0) | max | default(0)) %}                          # Object x max
    {% set purge_y_min = (all_points | map(attribute=1) | min | default(0)) %}                          # Object y min
    {% set purge_y_max = (all_points | map(attribute=1) | max | default(0)) %}                          # Object y max

    {% set purge_x_center = ([((purge_x_max + purge_x_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on X axis
    {% set purge_y_center = ([((purge_y_max + purge_y_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on Y axis

    {% set purge_x_origin = ([purge_x_min - purge_margin, 0] | max) %}                                  # Add margin to x min, compare to 0, and choose the larger
    {% set purge_y_origin = ([purge_y_min - purge_margin, 0] | max) %}                                  # Add margin to y min, compare to 0, and choose the larger

    # Calculate purge speed
    {% set purge_move_speed = (flow_rate / cross_section) * 60 | float %}

    {% if cross_section != 5 %}

        {action_respond_info("[Extruder] max_extrude_cross_section is not configured correctly, please set it to 5. Purge skipped.")}

    {% else %}

        {% if verbose_enable == True %}

        {action_respond_info("Moving filament tip {}mms".format(                                                                 
            (tip_distance),                                                                                      
        )) }
        {% endif %}

        {% if printer.firmware_retraction is defined %}
            {action_respond_info("KAMP purge is using firmware retraction.")}
        {% else %}
            {action_respond_info("KAMP purge is not using firmware retraction, it is recommended to configure it.")}
        {% endif %}

        {% if purge_y_origin > 0 %}
        
            {action_respond_info("KAMP purge starting at {}, {} and purging {}mm of filament, requested flow rate is {}mm/s3.".format(                                                                 
                (purge_x_center),
                (purge_y_origin),
                (purge_amount),
[... 23 lines cut ...]
            G1 X{purge_x_center + purge_amount} E{purge_amount} F{purge_move_speed}             # Purge line
            {RETRACT}                                                                           # Retract
            G92 E0                                                                              # Reset extruder distance
            M82                                                                                 # Absolute extrusion mode
            G0 Z{purge_height * 2} F{travel_speed}                                              # Z hop

        {% else %}                                                                              # If there's room on X, purge along Y axis to the left of print area
            
            G92 E0                                                                              # Reset extruder
            G0 F{travel_speed}                                                                  # Set travel speed
            G90                                                                                 # Absolute positioning
[... cut ...]
            {RETRACT}                                                                           # Retract
            G92 E0                                                                              # Reset extruder distance
            M82                                                                                 # Absolute extrusion mode
            G0 Z{purge_height * 2} F{travel_speed}                                              # Z hop

        {% endif %}
    
    {% endif %}
```

FiyaFly, 2023-04-20:
> @FiyaFly Hey, I really wanted to thank you for what your shared in terms of calculating the purge's movement speed. Using the max_cross_section value was pretty damn clever, and it seems to work nicely. That was a major pain for me to get it just right, and I think it's way better than it was before.
> 
> Here's the **DEVELOPMENT** macro if you're interested, for Line_Purge. It wont work without having a _KAMP_Settings macro (this is to greatly improve moonraker/setup-macro woes) but it's a nice sneak peak. 😉
> 
> ```
> [gcode_macro LINE_PURGE]
> description: A purge macro that adapts to be near your actual printed objects
> gcode:
>     # Get relevant printer params
>     {% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}
>     {% set cross_section = printer.configfile.settings.extruder.max_extrude_cross_section | float %}
>     
>     # Use firmware retraction if it is defined
>     {% if printer.firmware_retraction is defined %}
>         {% set RETRACT = G10 | string %}
>         {% set UNRETRACT = G11 | string %}
>     {% else %}
>         {% set RETRACT = 'G1 E-.5 F2100' | string %}
>         {% set UNRETRACT = 'G1 E.5 F2100' | string %}
>     {% endif %}
> 
>     # Get purge settings from _Kamp_Settings
>     {% set verbose_enable = printer["gcode_macro _KAMP_Settings"].verbose_enable | abs %}
>     {% set purge_height = printer["gcode_macro _KAMP_Settings"].purge_height | float %}
>     {% set tip_distance = printer["gcode_macro _KAMP_Settings"].tip_distance | float %}
>     {% set purge_margin = printer["gcode_macro _KAMP_Settings"].purge_margin | float %}
>     {% set purge_amount = printer["gcode_macro _KAMP_Settings"].purge_amount | float %}
>     {% set flow_rate = printer["gcode_macro _KAMP_Settings"].flow_rate | float %}
> 
> 
>     # Calculate purge origins and centers from objects
>     {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}    # Get all object points
>     {% set purge_x_min = (all_points | map(attribute=0) | min | default(0)) %}                          # Object x min
>     {% set purge_x_max = (all_points | map(attribute=0) | max | default(0)) %}                          # Object x max
>     {% set purge_y_min = (all_points | map(attribute=1) | min | default(0)) %}                          # Object y min
>     {% set purge_y_max = (all_points | map(attribute=1) | max | default(0)) %}                          # Object y max
> 
>     {% set purge_x_center = ([((purge_x_max + purge_x_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on X axis
>     {% set purge_y_center = ([((purge_y_max + purge_y_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on Y axis
> 
>     {% set purge_x_origin = ([purge_x_min - purge_margin, 0] | max) %}                                  # Add margin to x min, compare to 0, and choose the larger
>     {% set purge_y_origin = ([purge_y_min - purge_margin, 0] | max) %}                                  # Add margin to y min, compare to 0, and choose the larger
> 
>     # Calculate purge speed
>     {% set purge_move_speed = (flow_rate / cross_section) * 60 | float %}
> 
>     {% if cross_section != 5 %}
> 
>         {action_respond_info("[Extruder] max_extrude_cross_section is not configured correctly, please set it to 5. Purge skipped.")}
> 
>     {% else %}
> 
>         {% if verbose_enable == True %}
> 
>         {action_respond_info("Moving filament tip {}mms".format(                                                                 
>             (tip_distance),                                                                                      
>         )) }
>         {% endif %}
> 
>         {% if printer.firmware_retraction is defined %}
>             {action_respond_info("KAMP purge is using firmware retraction.")}
>         {% else %}
>             {action_respond_info("KAMP purge is not using firmware retraction, it is recommended to configure it.")}
>         {% endif %}
> 
>         {% if purge_y_origin > 0 %}
>         
>             {action_respond_info("KAMP purge starting at {}, {} and purging {}mm of filament, requested flow rate is {}mm/s3.".format(                                                                 
>                 (purge_x_center),
>                 (purge_y_origin),
[... 37 lines cut ...]
>             M83                                                                                 # Relative extrusion mode
>             G1 E{tip_distance} F{purge_move_speed}                                              # Move filament tip
>             G1 Y{purge_y_center + purge_amount} E{purge_amount} F{purge_move_speed}             # Purge line
>             {RETRACT}                                                                           # Retract
>             G92 E0                                                                              # Reset extruder distance
>             M82                                                                                 # Absolute extrusion mode
>             G0 Z{purge_height * 2} F{travel_speed}                                              # Z hop
> 
>         {% endif %}
>     
>     {% endif %}
> ```

I will definitely be checking this out this weekend and setting myself up to give this a try 😄 looks like the _KAMP_Settings macro should be pretty straightforward to setup.

[... cut ...]
2. ~~It would be possible to eliminate either `PURGE_AMOUNT` or `LINE_LENGTH`~~

Skimming through your code, looks like you've already solved Line Length as a dynamic 😄. In that case, it would be possible to make it even more dynamic. Really, just depends on what direction you'd want to go with the way various limits and parameters and handled. 

I plan to dive deeper this weekend, but these are just the things in the back of my head I can think of to try to help out with. Glad to jump in where you may want it or need it 😄

310weber, 2023-04-20:
> > Here's the **DEVELOPMENT** macro if you're interested, for Line_Purge. It wont work without having a _KAMP_Settings macro (this is to greatly improve moonraker/setup-macro woes) but it's a nice sneak peak. wink
> > ```
> > [gcode_macro LINE_PURGE]
> > description: A purge macro that adapts to be near your actual printed objects
> > gcode:
> >     # Get relevant printer params
> >     {% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}
> >     {% set cross_section = printer.configfile.settings.extruder.max_extrude_cross_section | float %}
> >     
> >     # Use firmware retraction if it is defined
> >     {% if printer.firmware_retraction is defined %}
> >         {% set RETRACT = G10 | string %}
> >         {% set UNRETRACT = G11 | string %}
> >     {% else %}
> >         {% set RETRACT = 'G1 E-.5 F2100' | string %}
> >         {% set UNRETRACT = 'G1 E.5 F2100' | string %}
> >     {% endif %}
> > 
> >     # Get purge settings from _Kamp_Settings
> >     {% set verbose_enable = printer["gcode_macro _KAMP_Settings"].verbose_enable | abs %}
> >     {% set purge_height = printer["gcode_macro _KAMP_Settings"].purge_height | float %}
> >     {% set tip_distance = printer["gcode_macro _KAMP_Settings"].tip_distance | float %}
> >     {% set purge_margin = printer["gcode_macro _KAMP_Settings"].purge_margin | float %}
> >     {% set purge_amount = printer["gcode_macro _KAMP_Settings"].purge_amount | float %}
> >     {% set flow_rate = printer["gcode_macro _KAMP_Settings"].flow_rate | float %}
> > 
> > 
> >     # Calculate purge origins and centers from objects
> >     {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}    # Get all object points
> >     {% set purge_x_min = (all_points | map(attribute=0) | min | default(0)) %}                          # Object x min
> >     {% set purge_x_max = (all_points | map(attribute=0) | max | default(0)) %}                          # Object x max
> >     {% set purge_y_min = (all_points | map(attribute=1) | min | default(0)) %}                          # Object y min
> >     {% set purge_y_max = (all_points | map(attribute=1) | max | default(0)) %}                          # Object y max
> > 
> >     {% set purge_x_center = ([((purge_x_max + purge_x_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on X axis
> >     {% set purge_y_center = ([((purge_y_max + purge_y_min) / 2) - (purge_amount / 2), 0] | max) %}      # Create center point of purge line relative to print on Y axis
> > 
> >     {% set purge_x_origin = ([purge_x_min - purge_margin, 0] | max) %}                                  # Add margin to x min, compare to 0, and choose the larger
> >     {% set purge_y_origin = ([purge_y_min - purge_margin, 0] | max) %}                                  # Add margin to y min, compare to 0, and choose the larger
> > 
> >     # Calculate purge speed
> >     {% set purge_move_speed = (flow_rate / cross_section) * 60 | float %}
> > 
> >     {% if cross_section != 5 %}
> > 
> >         {action_respond_info("[Extruder] max_extrude_cross_section is not configured correctly, please set it to 5. Purge skipped.")}
> > 
> >     {% else %}
> > 
> >         {% if verbose_enable == True %}
> > 
> >         {action_respond_info("Moving filament tip {}mms".format(                                                                 
> >             (tip_distance),                                                                                      
> >         )) }
> >         {% endif %}
> > 
> >         {% if printer.firmware_retraction is defined %}
> >             {action_respond_info("KAMP purge is using firmware retraction.")}
> >         {% else %}
> >             {action_respond_info("KAMP purge is not using firmware retraction, it is recommended to configure it.")}
> >         {% endif %}
> > 
> >         {% if purge_y_origin > 0 %}
> >         
> >             {action_respond_info("KAMP purge starting at {}, {} and purging {}mm of filament, requested flow rate is {}mm/s3.".format(                                                                 
> >                 (purge_x_center),
> >                 (purge_y_origin),
> >                 (purge_amount),
> >                 (flow_rate),
> >             )) }
[... 33 lines cut ...]
> >             G0 Z{purge_height}                                                                  # Move to purge Z height
> >             M83                                                                                 # Relative extrusion mode
> >             G1 E{tip_distance} F{purge_move_speed}                                              # Move filament tip
> >             G1 Y{purge_y_center + purge_amount} E{purge_amount} F{purge_move_speed}             # Purge line
> >             {RETRACT}                                                                           # Retract
> >             G92 E0                                                                              # Reset extruder distance
> >             M82                                                                                 # Absolute extrusion mode
> >             G0 Z{purge_height * 2} F{travel_speed}                                              # Z hop
> > 
> >         {% endif %}
> >     
> >     {% endif %}
> > ```

[... cut ...]
One other possible optimization would be to calculate `purge_height` based on `cross_section` rather than having to configure it in KAMP settings.  I found in my testing that keeping the ratio of purge_width:purge_height to roughly 1.5 worked well.  i.e. `cross_section = purge_height * 1.5*purge_height`.  With that in mind you could use:

    purge_height = (cross_section/1.5)**0.5

Otherwise I think it looks great.  I like how simple the error checking is to simply print on the opposite edge if there's no room on Y.

closed 2023-07-06 by kyleisah