kyleisah/Klipper-Adaptive-Meshing-Purging #39: Adaptive Meshing Always in Front-Left Corner

chrison999, 2023-03-04:
Hi!  I have followed the instructions for installing adaptive meshing on mainsail, and it seems to be working but with one problem.  It always does the mesh in the front-left corner of the bed even though the object being printed is centred on the bed.  The dimensions of the mesh seems to be right, but the location is wrong.

kyleisah, 2023-03-04:
Hm, that's odd. Got a sliced gcode file for me?

chrison999, 2023-03-07:
Sorry for the delay.  Been kinda sick the last few days.  Anyway, heres a gcode file I created today using prusa 2.5.0.  Will send you a klippy.log shortly.

Thanks for your help!  Chris
[turtle_with_moving_legs_prusa.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/10913391/turtle_with_moving_legs_prusa.zip)

chrison999, 2023-03-08:
Here's a copy of my recent klippy.log
[klippy (1).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/10923155/klippy.1.log)

kyleisah, 2023-04-02:
Hello, sorry for the delay, I don't know why this would be forcing the meshed area to be in a corner of the bed. I want to say you don't have your axes defined correctly or something. KAMP has been around for a while and this is a totally new issue.

chrison999, 2023-04-02:
Ok.  Thanks for the reply.  I’ll double check my configs for any errors.

kyleisah, 2023-04-13:
Because I cannot reproduce the issue, I'm going to close this for now.

closed 2023-04-13 by kyleisah