kyleisah/Klipper-Adaptive-Meshing-Purging #40: line purge overlapped with brim

uglyfish333, 2023-03-06:
Basically what the title says. Printed something with a large brim (like 20mm if I recall correctly) and the purge line was overlapped with the brim. Seems maybe the object spacing/size didn't include the brim. Not sure if this was a problem with superslicer, processing in klipper, or what.

kyleisah, 2023-03-06:
I don't think any slicers label the brim as an object, so unless that happens, you'll have to compensate for it another way. I suggest adjusting 'distance_to_object_y' for Line_Purge

uglyfish333, 2023-03-06:
Good to know. Will just need to remember to adjust/turn off adaptive in the rate occasion of using a brim. It maybe see if brim width is a parameter the slicer can pass to the start code quick can add it to the normal offset before passing that to line purge. If I figure it out I can post back. Great work on this project by the way it's much appreciated.

closed 2023-03-08 by kyleisah