kyleisah/Klipper-Adaptive-Meshing-Purging #41: Marco Parameters

KiwiChrist, 2023-03-07:
1. Whatever fingure I enter into the 'setup purge line' parameters marco and 'send' nothing changes. Am I using it correctly?
2. 
[image]

3. The 'purge line.cfg' varaiables do not seem to do adjust the purge line print when altering them. When adjusting should they not adjust the purge line print? 

[image]

Typically I have followed the install instructions and altered the variables as nessary but not receiving any results. I am wondering if I am using this correctly.

KiwiChrist, 2023-03-07:
I have worked it out reading the instructions again and tinkering.

closed 2023-03-07 by KiwiChrist

kyleisah, 2023-03-08:
Those parameters do not display changes during runtime. Once you set them, they are set until you restart. This is how it is for those using the moonraker updater installation method.