kyleisah/Klipper-Adaptive-Meshing-Purging #42: Jinja error when trying to print Gcode from Cura 5.3

gillesguillemin, 2023-03-07:
I'm getting this error in Klipper's log:
```
!! Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'
```
Where it gets weird, is that this happens with Gcode generated by Cura 5.3. No such problem with 5.2 (or even 5.3 Christmas alpha). I saw this too with the recent 5.3 beta but thought that was some kink needing ironing out but now 5.3 final is out and the problem remains.

gillesguillemin, 2023-03-07:
Oh and this happens whichever model I throw at it.

kg6ebj, 2023-03-08:
I backdated Cura to 5.2.2 to resolve the issue and have it co-installed with 5.3.0, which will not work because of the error.

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

kyleisah, 2023-03-08:
@kg6ebj 
[image]
Seems like cura for whatever reason is generating an object named "nomesh" without a polygon definition. This isn't a problem with KAMP, but a problem with Cura, or how you have Cura 5.3 configured

closed 2023-03-08 by kyleisah

gillesguillemin, 2023-03-08:
And it had been flagged on the Cura repo actually. I missed that yesterday when I opened this one.

https://github.com/Ultimaker/Cura/issues/14679

kg6ebj, 2023-03-08:
> @kg6ebj
> [image]
> Seems like cura for whatever reason is generating an object named "nomesh" without a polygon definition. This isn't a problem with KAMP, but a problem with Cura, or how you have Cura 5.3 configured

There is a typo in Cura's code.
If you just replace NOMESH

Cura -> Menu -> Extensions -> Post Processing -> Modify GCode

Then Add a Script -> Search and Replace
Search is NOMESH
Replace is NONMESH

It resolves this issue