kyleisah/Klipper-Adaptive-Meshing-Purging #44: Adaptive mesh only clamping on one AXIS

DrDrache, 2023-03-13:
Printer : Voron 2.4 - recore A6 - Voron Tap
slicer : PS 2.6-alpha5

as the title states, the adaptive mesh is only clamping on a single axis - as seen here :

Object bounds, clamped to the bed_mesh: (40.0, 40.0), (310.0, 310.0)
0 object points, clamping to bed mesh [(40.0, 40.0) (310.0, 310.0)]

attached is printer.cfg - and klippy.log

I don't know if this was because of a move to PS - or a move to newer KAMP (was using a "older" version before I updated to the current) or both?

I attempted to add the M117 line - but not sure I did that right. 
start print - and screenies for people who don't have the time to read/extract the configs :
[include Adaptive_Mesh.cfg] - direct copy from this GitHub
[include Voron_Purge.cfg] - same as above

[image]
[image]
[image]
[image]
[image]

[v24-configs.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/10958256/v24-configs.zip)

kyleisah, 2023-04-26:
0 object points means that Klipper is not detecting any excludable objects, something isn't correct with your setup. If you have gone through the setup correctly, and it's still not working, you likely need to repair your moonraker installation because the object processor is not working correctly.

AxMod3DPrint, 2023-04-27:
Random question @DrDrache are you running a probe such as Klicky or the like? I've 2 printers running KAMP and after a bit of checking with someone on the Voron Discord and a fresh install of Klipper/Moonraker/Fluidd, it seems something isn't processing properly when running a Klicky probe but works fine with an inductive. This is both uploading manually and via Superslicer.

kyleisah, 2023-04-28:
@AxMod3DPrint That has nothing to do with KAMP. Inductive uses the **default** bed_mesh_calibrate command typically. Klicky has it's own version of bed_mesh_calibrate for **attaching the probe**. Both of which, KAMP would replace by overwriting the default macro, or commenting out klicky's macro, and then calling the attach probe command from KAMP's `attach_macro` variable.

AxMod3DPrint, 2023-04-28:
I realise that. Something is causing KAMP not to clamp to the parts in the GCode. The only difference between the two I have set up is Klicky. They were both installed the same way (kiauh) and everything is sliced the same way with the same settings. This is why I asked. I'm only trying to eliminate potential factors that could be causing this. 

Moonraker _is_ processing but Klipper isn't picking up the objects (exclude object button is greyed out) which in turn is stopping KAMP from clamping.

kyleisah, 2023-04-28:
> I realise that. Something is causing KAMP not to clamp to the parts in the GCode. The only difference between the two I have set up is Klicky. They were both installed the same way (kiauh) and [...]

See: https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/issues/44#issuecomment-1522848328

AxMod3DPrint, 2023-04-28:
Nope. eliminated that variable, did a fresh install using kiauh, full SD Card wipe, which the other printer was installed with and works fine.

closed 2023-07-06 by kyleisah