kyleisah/Klipper-Adaptive-Meshing-Purging #46: Polygon attribute error running bed_mesh_calibrate

EklDuderino, 2023-03-16:
Having an issue when the gcode_macro BED_MESH_CALIBRATE runs. It returns the attached error. Sliced in cura 5.3.0. Looks like something is not getting passed along properly from Cura to the printer regarding exclude_object, but I am unsure how to resolve this. Thank you.
[image]

EklDuderino, 2023-03-16:
Found solution to this issue. Seems to be 5.3 Cura bug. https://github.com/Ultimaker/Cura/issues/14679

kyleisah, 2023-04-13:
Thanks for bringing this to my attention! Seeing as it is a Cura bug, I am going to be closing this issue for now.

closed 2023-04-13 by kyleisah