kyleisah/Klipper-Adaptive-Meshing-Purging #47: Adaptive Purge location ignores brim

dlettiere, 2023-03-16:
Started a print with 30mm brim (was having draft issues). Voron purge location was within the brim. I didn't realize until after, luckily the Ender2 can flex and was just driving right over the "purge pellets".

I understand I can adjust the location for the purge in the config, and also that a 30mm brim is a bit nuts, but figure I should share the issue anyhow!

Thanks for the awesome tool!!

dmaxwellNEU, 2023-04-05:
I have just observed the exact same issue with the line purge. I need a 20mm brim for polypropylene printing, and the purge line would have been right inside the brim if I hadn't pulled it off immediately. I hope this issue is fixed because I love the adaptive purging.

dmaxwellNEU, 2023-04-05:
> > I have just observed the exact same issue with the line purge. I need a 20mm brim for polypropylene printing, and the purge line would have been right inside the brim if I hadn't pulled it off immediately. I hope this issue is fixed because I love the adaptive purging.
> 
> You can adjust the "variable_distance_to_object_x: 10" and "variable_distance_to_object_y: 10" in the Voron_Purge.cfg to be further away for large brim prints, I set mine to 20mm for both and it works just as well!!

I see, thanks for the response! The only issue is that this will be irritating to remember to change for each print when there is a brim and when there isn't a brim. I'm hoping that the devs could just readjust the code to include the brim in the base functionality so we don't have to change this every time. My recommendation would be that the variables distance x and y would be applied to the brim, not the object without the brim.

dlettiere, 2023-04-05:
Its possible to pass the brim width via start G-code in SuperSlicer. I have added  BRIM={brim_width} to mine. Apparently you can output almost any value this way.

- PREP_PRINT EXTRUDER={first_layer_temperature[initial_extruder] + extruder_temperature_offset[initial_extruder]} BED={first_layer_bed_temperature} CHAMBER={chamber_temperature} FILAMENT={filament_type} COUNT={total_layer_count} TOOLS={total_toolchanges} NUM=1 BRIM={brim_width}

gave me a g-code output of 

- PREP_PRINT EXTRUDER=240 BED=70 CHAMBER=0 FILAMENT=PETG COUNT=8 TOOLS=0 NUM=1 BRIM=10 

it should be possible to pass this to the macro to compensate for the brim width like "variable_distance_to_object_x: 10+ passedBRIMvalue"

I have a huge blind spot with passive variable in klipper otherwise I'd prob take a crack at it.

kyleisah, 2023-04-08:
Hey ya'll, so sorry for my absence, I can see this becoming quite annoying for those who need large brims on certain prints. I will see what I can do about passing some sort of custom variable, or tricking the macro into adjusting the margin according to the brim value, that doesn't sound terribly hard. Thank you!

dmaxwellNEU, 2023-04-08:
> Hey ya'll, so sorry for my absence, I can see this becoming quite annoying for those who need large brims on certain prints. I will see what I can do about passing some sort of custom variable, or [...]

That would be amazing! Thanks!

kyleisah, 2023-04-09:
@dlettiere @dmaxwellNEU 

I think if you add this to your slicer *before* print_start in the start gcode section, it might work:

SET_GCODE_VARIABLE MACRO=VORON_PURGE VARIABLE=distance_to_object_x VALUE={brim_width}
SET_GCODE_VARIABLE MACRO=VORON_PURGE VARIABLE=distance_to_object_y VALUE={brim_width}

Please test this and let me know. This should set the distance value to the brim's width, hopefully!

dlettiere, 2023-04-09:
I am still using the older version of KAMP so I added:

gcode:
    {% set distance_to_object_x = distance_to_object_x + printer["gcode_macro PREP_PRINT"].brim_width %}
    {% set distance_to_object_y = distance_to_object_y + printer["gcode_macro PREP_PRINT"].brim_width %}

into my AdaptivePurge.cfg

In my prep print macro I am reading the passed brim value from the slicer start gcode (or 0 default). 
gcode:
	 {% set BRIMWIDTH = params.BRIM |default(0.0)| float %}
	SET_GCODE_VARIABLE MACRO=PREP_PRINT VARIABLE=brim_width VALUE={BRIMWIDTH}

It works exactly as expected and gave me a good opportunity to learn up on variables.

I will give your suggestion a try next time I'm at my printers. Would this put the center of the "purgelet" on the edge of the brim or would the purgelet size be factored in?

kyleisah, 2023-04-09:
@dlettiere If memory serves, the "purgelets" origin should be at the top of the middle purgelet, or the top right corner of the logo. What I shared above should do exactly what you have made for your PREP_PRINT macro, except it will not **add** the brim_width to the distance variables, because it's passing values from your slicer to the macro, rather that from the slicer to your config like you have it. The way you have it is pretty smart given our current constraints. I could see about adding some kind of modifier somehow, but that's gonna be a little bit of work.

I could just add a BRIM variable or something, but i'd like to maybe call it CUSTOM or something like that, and work out ways to combine values with some gcode commands.

kyleisah, 2023-04-09:
Another possibility is to work out a way for the slicer to pass the brim as an object, so KAMP can parse it just like any other thing

kyleisah, 2023-04-20:
@dlettiere The next release of KAMP will have a hidden macro for advanced configuration and debugging. I'm not 100% sure how this will work with *adding* the brim value to the purge_margin value, but it's a step!

[image]

closed 2023-07-06 by kyleisah