kyleisah/Klipper-Adaptive-Meshing-Purging #48: purge line very close to print

cperricos, 2023-03-27:
hello,
I like the adaptive purge, but the purge line is always about a mm from the actual print. i have tried with a nymber of variables and tthey don't see to have any effect. my code section is:

[gcode_macro LINE_PURGE]
description: A purge macro that adapts to be near your actual printed objects

variable_adaptive_enable: True      # Change to False if you'd like the purge to be in the same spot every print
variable_z_height: 0.4              # Height above the bed to purge
variable_tip_distance: 10            # Overall desired length of purge line in millimeters, around 1/5th of X axis length is a good starting value
variable_purge_amount: 20           # Amount of filament in millimeters to purge
variable_line_length: 30            # Overall desired length of purge line in millimeters, around 1/5th of X axis length is a good starting value
variable_flow_rate: 8              # Desired flow rate in mm3/s (Around 12 for standard flow hotends, around 24 for high flow hotends)
variable_x_default: 50              # Default X location to purge. If adaptive_enable is True, this is overwritten
variable_y_default: 50              # Default Y location to purge. If adaptive_enable is True, this is overwritten
variable_size: 10
variable_distance_to_object_x: 40   # x distance in millimeters away from the print area for purging. Must be less than or equal to y_default if adaptive_enable is False
variable_distance_to_object_y: 40  # Y distance in millimeters away from the print area for purging. Must be less than or equal to y_default if adaptive_enable is False
variable_travel_speed: 300

any ideas?

many thanks

kyleisah, 2023-04-13:
Hello! So sorry to have gotten back to you after so long. Have you had any luck figuring this out? I've been pretty busy lately unfortunately. Your configuration looks like it should be working fine. Can you perhaps give me a little more information, or a klippy.log file?

cperricos, 2023-04-13:
Hello,
  
Many thanks for the response. I have gone back to regular purging with adaptive meshing and it works perfectly now so will leave it that way.   
  

  
Thanks

kyleisah, 2023-04-20:
@cperricos just following up with you, the next release of KAMP will have a central "settings" macro that will allow you to define a purge margin, so you will always have a set distance from the print to the purge. 😄 

[image]

cperricos, 2023-04-20:
Ah that’s sounds excellent -    thanks

closed 2023-07-06 by kyleisah