kyleisah/Klipper-Adaptive-Meshing-Purging #50: Can I use this with sonic pad?

Savijen, 2023-04-01:
Just ordered one and this project sounds great. Will I be able to use it or is it just for native klipper installs?

kyleisah, 2023-04-02:
I actually don't know for sure, but I imagine it does. If you are able to `ssh` into the sonic pad, I imagine you can do just about anything.

kyleisah, 2023-04-09:
@Savijen any updates on this? You've got me curious 😅

Savijen, 2023-04-09:
I have not tried. You can get root on it and ssh but apparently it's very limited what you can do. Since I just received it and am still getting things tweaked to perfection, I have not had a chance to play with this....
 Yet...

JamesH1978, 2023-04-09:
no, sonic pad has no exclude/cancel print functionality

Thanks
James

kyleisah, 2023-04-09:
> no, sonic pad has no exclude/cancel print functionality
> 
> 
> 
> Thanks
> 
> James

Does sonic pad run a fork of an old version of klipper or something?

JamesH1978, 2023-04-09:
yes they forked an old version of klipper, then closed their source code.

kyleisah, 2023-04-13:
I am going to close this issue for now, as it is not a KAMP problem. I am still curious if it works on the sonic pad, but from the little information I can find, I can't confirm anything. Thank you!

closed 2023-04-13 by kyleisah