kyleisah/Klipper-Adaptive-Meshing-Purging #51: BED_MESH_CALIBRATE Parameter "PROFILE" not functional - with fix

weekendclimber, 2023-04-05:
I do a bed mesh on each print. I like to save off bed meshes with filenames for future reference. I've got this in my START_PRINT macro:

```
    {% if params.ST is not defined %}
        {% set ST = config.extruder.pressure_advance_smooth_time %}
    {% endif %}

    BED_MESH_CLEAR
    SETUP_KAMP_MESHING DISPLAY_PARAMETERS=1 LED_ENABLE=1 FUZZ_ENABLE=1
    BED_MESH_CALIBRATE PROFILE={FNAME} ;Calibrate mesh
    BED_MESH_PROFILE LOAD={FNAME}
```

The "PROFILE" parament does not get sent to the system level BED_MESH_CALIBRATE gcode macro because it is not defined in the call here (line 121 of Adaptive_Mesh.cfg):

`_BED_MESH_CALIBRATE mesh_min={x_min},{y_min} mesh_max={x_max},{y_max} ALGORITHM={algorithm} PROBE_COUNT={points_x},{points_y} RELATIVE_REFERENCE_INDEX={ref_index}`

I've updated that line to pass the "PROFILE" parameter and if not present, set it to 'default'. I also added a line above so that they are now:

```
    {% set profile = params.PROFILE|default('default')|string %}
    _BED_MESH_CALIBRATE mesh_min={x_min},{y_min} mesh_max={x_max},{y_max} ALGORITHM={algorithm} PROBE_COUNT={points_x},{points_y} RELATIVE_REFERENCE_INDEX={ref_index} PROFILE={profile}
```

kyleisah, 2023-04-08:
Hey there! I appreciate you taking the time to let me know about this, I think it would be handy to be able to save profiles for later referencing. Unfortunately, KAMP is undergoing a fairly large overhaul in dev right now, so I can't push this right away, but I will be sure to include a profile saving method when that is released. Thank you!

kyleisah, 2023-04-09:
Seems like profiles are pretty easy to work with. I'm pretty sure that from within fluidd or mainsail's "heightmap" section, you can work with profiles to save them. I don't know much about it, because I don't use these functions, but that's a possibility.

There are also provisions within Klipper for saving bed meshes using commands. That information can be found here: https://www.klipper3d.org/Bed_Mesh.html#profiles

`BED_MESH_PROFILE SAVE=<name> LOAD=<name> REMOVE=<name>` is the command and parameters.

If you wanted to make it a bit easier, you could make yourself a helper macro.

Something like:

```jinja
[gcode_macro SAVE_MESH]
description: Saves your current mesh to your printer.cfg with the name "default".
gcode:
        BED_MESH_PROFILE SAVE=default
        {action_respond_info("Created mesh profile named "default". Use SAVE_CONFIG to save it.")}
```

You could then add that to the bottom of your PRINT_END macro or something like that, so it will save the mesh when the print finishes. 😄

kyleisah, 2023-04-09:
To add to this, it looks like the name `default` for meshes is reserved, because that is the default name given to any probed mesh during runtime. I'm kind of leaning toward keeping this out of KAMP, because probing an adapted mesh before every print is kind of the ethos of KAMP. However, we both learned a little bit about mesh profiles. 😄

kyleisah, 2023-04-13:
I'm going to close this issue for now, because Klipper already has a means of saving meshes through the front-end. Thank you!

closed 2023-04-13 by kyleisah