kyleisah/Klipper-Adaptive-Meshing-Purging #52: Failure: bed_mesh: min/max points too close together

tommy536, 2023-04-08:
Since using KAMP, no bed_mesh is possible. It can be, that the print part is realy smal. 

What can i do to get it work?

kyleisah, 2023-04-08:
Can you send me a klipper log file? I could've sworn we had this issue before and fixed it. Thanks!

kyleisah, 2023-04-09:
Oh wow, seems like (at least with my machine) an object has to be smaller than 2mm to throw that error 😅. I have an idea of how this can be fixed though. Until that time comes, I guess you'll just have to put more objects on the bed, or set the margin_size to something like 5 or 10.

leezus-nz, 2023-04-11:
FYI, I'm getting that error too.

Happens no matter what size the print is. Using Orca Slicer.

[klippy.log.2023-04-07.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11200095/klippy.log.2023-04-07.txt)

kyleisah, 2023-04-11:
@leezus-nz Can you send me a sliced gcode file? I've only just heard of orca slicer and I'm curious how it's labeling object. Thank you!

leezus-nz, 2023-04-11:
@kyleisah sure thing.

It's a good slicer. Based on Bambu. It has been working in the past, I haven't tried rolling back to an older version. Strange thing is that I have managed to get it to work with this current version, but not normally. I've just disabled KAMP for now. Feels so painful watching my printer mesh the entire bed.

[[o]_stealthburner_LED_carrier_ABS_8m27s.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11205622/o._stealthburner_LED_carrier_ABS_8m27s.txt)

jaakko000, 2023-04-12:
[V1.5 Train-Whistle-Low-Pitch_ASA_38m14s.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11216341/V1.5.Train-Whistle-Low-Pitch_ASA_38m14s.txt)
[V1.6.1 Train-Whistle-Low-Pitch_ASA_37m26s.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11216342/V1.6.1.Train-Whistle-Low-Pitch_ASA_37m26s.txt)

Pitching in with mine. I dont have a klippy log sorry. KAMP works fine on V1.5 and previous versions. On V1.6 Orcaslicer messed with object_exclusion and V1.6.1 quickly hotfixed some issue with it. KAMP doesnt work on V1.6 or 1.6.1 just gives the error of title and says "[way too many] object points found"
[image]

Hope you are able to fix this

kyleisah, 2023-04-13:
@jaakko000 just getting back to this now, and thank you for the gcode files, very helpful! 
[image]

It seems, for some strange reason, Orcaslicer is defining the points incorrectly for your objects. They're reasonable right up until you start getting into *-900* or so on the Y axis. 😂 I think there's a bug with orcaslicer.

kyleisah, 2023-04-13:
This is also apparent in the gcode file you sent over to me 
[image]

kyleisah, 2023-04-13:
@jaakko000 I am looking into your files now, just a moment! Sorry for the ghost ping, if you got that. 😅

kyleisah, 2023-04-13:
@jaakko000 Alright, so it looks like in V1.5, you disabled (or had unchecked) object labeling, so no objects were getting labeled by the slicer, which is why KAMP was able to work. If no points are passed, a full mesh is ran. But it looks like in the V1.6 gcode, you had object labeling enabled (obviously!) but it seems like OrcaSlicer is adding some weird offset to the machine size. The polygonal coordinates that define the shape of the object have some sort of weird offset being applied to them somehow from Orcaslicer. I dont imagine a whistle being larger than 300mm, yet orcaslicer is outputting it as such.

have a look here:

[image]

This is saying the center of the object (likely the center of your printer's buildplate) is located at ~393,130. So your printer's total buildsize would have to be something like 800, 300, which is utterly insane for hobbyists like us lol.

kyleisah, 2023-04-13:
Seems to have been reported here: https://github.com/SoftFever/OrcaSlicer/issues/668. I've commented letting everyone know I'm following along.

Thanks for bringing this to my attention. Unfortunately, I don't think I'll be able to help the devs of Orcaslicer fix their code, so if you want meshing to work fine, I would just turn off labeling objects for now.

leezus-nz, 2023-04-13:
Thanks for your help.

jaakko000, 2023-04-13:
https://github.com/SoftFever/OrcaSlicer/issues/668

Fixed in dev branch according to developer, if i understand correctly that means it should be working in next release. Interesting info in the last commit of that thread. Thank you very much @kyleisah

tommy536, 2023-04-16:
> Oh wow, seems like (at least with my machine) an object has to be smaller than 2mm to throw that error 😅. I have an idea of how this can be fixed though. Until that time comes, I guess you'll just have to put more objects on the bed, or set the margin_size to something like 5 or 10.

Seems, that the combination of Orca Slicer and KAMP are not compatible. Bcs. i am using Orca too. I will try to disabling labeling of the objects.

kyleisah, 2023-04-16:
@tommy536 yes, because OrcaSlicer is (currently) defining objects incorrectly. It seems there is already a fix in their dev branch.

kyleisah, 2023-04-20:
Looks like orcaslicer has patched this up! I will close this now. Happy KAMPing! 😄

tommy536, 2023-04-21:
Yes! With the new Orca Slicer Version 1.6.2 adaptive Mesh works now! Thank u!

closed 2023-04-26 by kyleisah