kyleisah/Klipper-Adaptive-Meshing-Purging #53: KAMP + Auto_Z does not have Auto_Z probing KAMP's RRI

sdooweloc, 2023-04-09:
I recently installed KAMP and am already using Calibrate_Z. 
KAMP install went great but I have noticed that the Calibrate_Z macro is not honoring KAMP's RRI causing offcenter prints to not probe the center of the print area and instead probe the default center of the bed. 

I have tried doing Calibrate_Z both before and after the Bed_Mesh_Calibrate but both ways have this issue. 
I have included my Print_Start below as well as my Calibrate_Z config and the I think relevant section of Calibrate_z's .py and am using the current state of the KAMP repo. 

Any and all assistance is appreciated and if you need more information please just let me know.

`[gcode_macro PRINT_START]
#   Use PRINT_START for the slicer starting script - please customise for your slicer of choice
gcode:
    # Parameters
    {% set bedtemp = params.BED|int %}
    {% set hotendtemp = params.HOTEND|int %}

    SETUP_KAMP_MESHING DISPLAY_PARAMETERS=1 LED_ENABLE=1 FUZZ_ENABLE=1 probe_dock_enable=1
    SETUP_LINE_PURGE DISPLAY_PARAMETERS=1 ADAPTIVE_ENABLE=1
    
    SET_PIN PIN=caselight VALUE=0.5
    STATUS_HOMING
    G28
    STATUS_HEATING
    M190 S{bedtemp}

    M109 S180
    G32                            ; home all axes

    STATUS_MESHING

    BED_MESH_CLEAR
    BED_MESH_CALIBRATE

    CALIBRATE_Z
    SET_GCODE_OFFSET Z_ADJUST={params.Z_ADJUST|default(0.0)|float} MOVE=1
    G0 X40 Y300 F3600
    M109 S{hotendtemp}
    SONG_STARWARS_IMP

    STATUS_PRINTING`
    
    `[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    {% set bed_position = params.BED_POSITION|default('None') %}

    G28                         # can also be a conditional homing macro
    M117 Z-Calibration..
    ATTACH_PROBE                # a macro for fetching the probe first
    G1 X206 F8000 # to avoid the dock

    {% if bed_position != 'None' %}
      RESPOND MSG="Z calibration: a mesh is computed and ready, probing mesh center: {mesh_center}"
      BASE_CALIBRATE_Z BED_POSITION={bed_position}
    {% else %}
      RESPOND MSG="Z calibration: no mesh computed, probing default point"
      BASE_CALIBRATE_Z
    {% endif %}

    DOCK_PROBE                # and parking it afterwards (or DOCK_PROBE in klicky macros)
    M117`
    
    
    https://github.com/protoloft/klipper_z_calibration/blob/f212b409cd1728b8edaabc7ec0f6144542dd5356/z_calibration.py#L109
    ```py
    [gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    {% set bed_position = params.BED_POSITION|default('None') %}

    G28                         # can also be a conditional homing macro
    M117 Z-Calibration..
    ATTACH_PROBE                # a macro for fetching the probe first
    G1 X206 F8000 # to avoid the dock

    {% if bed_position != 'None' %}
      RESPOND MSG="Z calibration: a mesh is computed and ready, probing mesh center: {mesh_center}"
      BASE_CALIBRATE_Z BED_POSITION={bed_position}
    {% else %}
      RESPOND MSG="Z calibration: no mesh computed, probing default point"
      BASE_CALIBRATE_Z
    {% endif %}

    DOCK_PROBE                # and parking it afterwards (or DOCK_PROBE in klicky macros)
    M117
    ```

kyleisah, 2023-04-09:
Thank you for this! I will look into when I've got some time to. Likely when I get my trident back up and running as I'll be using a Klicky probe again. 😄

tijnfaas, 2023-04-29:
I have the same issue. When using auto_z with Klicky, the nozzle is way too high from the bed.
Is there anything I can do to help you diagnose this @kyleisah, before you have your trident back up?

closed 2023-07-06 by kyleisah

big3dworld, 2023-07-10:
This was closed but any solution came about? I'm currently running the same issue, I'm printing way too high after switching over

tijnfaas, 2023-07-10:
No, I am still working with the workaround of using an older version. That works so well that I didn’t even checked newer ones :)

big3dworld, 2023-07-10:
Awesome thanks for contributing your time to this! My printer will be out of commission until this gets resolved so if you need any testers/developers let me know :)

kyleisah, 2023-07-10:
You could also call BED_MESH_CALIBRATE from within the Z calibration custom gcode configuration, that's how I did it on my trident so I wouldn't pick up and dock the probe 3 different times. Maybe that's the ticket.

big3dworld, 2023-07-11:
Ok, I will try all those things, thank you for the guidance :) anddd yea you are right I could not use it but im weird like that I guess. I still have my switchwire I can use to print stuff so its not _urgent_ for me. 

I realized that the purge line gets printed perfectly! 

Right now my KAMP is imported before Klicky.
=====================================

So, I changed 

```
[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    M117 Z-Calibration..
    SET_GCODE_OFFSET Z=0
    # 
    BASE_CALIBRATE_Z
    M117
```

To 
```
[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    M117 Z-Calibration..
    SET_GCODE_OFFSET Z=0
    BED_MESH_CLEAR
    BED_MESH_CALIBRATE
    # 
    BASE_CALIBRATE_Z
    # SET_GCODE_OFFSET Z_ADJUST=2.00 MOVE=1
    M117
```

And my print start now looks like this 

```
[gcode_macro PRINT_START]
gcode:
    SAVE_GCODE_STATE NAME=PRINT_START_STATE
    G28
    G90                            ; absolute positioning
    G1 Z20 F3000                   ; move nozzle away from bed
    # M117 Heating bed
    # M190 S{ params.BED }
    # M117 Heating nozzle
    # M109 S{ params.EXTRUDER }  ; Wait for nozzle to get to target temperature
    M117 Heating bed
    
    M140 S{ params.BED } ; Start heating bed without waiting
    M104 S{ params.EXTRUDER }
    M190 S{ params.BED } ; Wait for bed to reach temp before proceeding
    M109 S{ params.EXTRUDER }  ; Wait for nozzle to get to target temperature

    G1 E45 F1800                  ; Push what we retracted 
    CLEAN_NOZZLE
    G32                            ; home all axes
    
    CALIBRATE_Z
    
    M117 Heating nozzle
    {% if params.FILAMENT == "PLA" %}
      M117 Printing PLA! 
      # SET_GCODE_OFFSET Z_ADJUST=+0.20 ; For PLA
    {% endif %}
    LINE_PURGE
    RESTORE_GCODE_STATE NAME=PRINT_START_STATE

```

Making the changes nothing got better. I did notice however that the Z_Offset was not being applied, Maybe this is a auto Z issue and not KAMP

big3dworld, 2023-07-11:
OK FORGET ABOUT ME!!! As always, the user is the problem 😶 . After updating Z-Calibration I had a `SAVE_GCODE_STATE` and `RESTORE_GCODE_STATE` which messes it up. Sorry for hijacking thread, just putting this here in case someone stumbles into this in the future.