kyleisah/Klipper-Adaptive-Meshing-Purging #55: Adaptive Mesh not accounting for probe offset

AxMod3DPrint, 2023-04-15:
Hi,

I've just installed this and when probing, it doesn't seem to account my X offset of -40.3, I'm not too sure about Y due to probing being in the middle of the bed. 

Instead of having the probe go where the print is, it is probing where the nozzle is, with an effective 0 offset.  Label objects is on in the slicer (SuperSlicer) and Moonraker and my printer cfg files are set up correctly. 

See [here](https://youtu.be/giCS-HMuuUw) for a video of the issue

kyleisah, 2023-04-16:
I think there's something wrong with your [probe] config. KAMP doesn't change offsets or anything like that, it just changes the min/max or the default bed mesh and adjusts points. Can you share your [probe] section?

kyleisah, 2023-04-16:
`// 4 object points, clamping to bed mesh [(-61.7, -9.5) (120.0, 185.0)]` this is also strange. The min point here is -61.7

AxMod3DPrint, 2023-04-16:
> I think there's something wrong with your [probe] config. KAMP doesn't change offsets or anything like that, it just changes the min/max or the default bed mesh and adjusts points. Can you share your [probe] section?

Probe section below:

```
[probe]
pin = ^!ar18
x_offset = -40.3
y_offset = -9.5
#z_offset = 0.12
samples: 3
```

Z offset is set using PROBE_CALIBRATE

AxMod3DPrint, 2023-04-16:
> `// 4 object points, clamping to bed mesh [(-61.7, -9.5) (120.0, 185.0)]` this is also strange. The min point here is -61.7

Think I may have found where the -61.7 has come from:

```
[bed_mesh]
speed = 100
horizontal_move_z = 5
mesh_min = -61.7,-9.5
mesh_max = 120,185
probe_count = 5,5
fade_start: 1
fade_end: 10
fade_target: 0
```

I take a look at that, but I think I needed that to get the probe to probe at the edge of the bed.

kyleisah, 2023-04-16:
I think you should maybe look into defining the origin (0,0) of your machine correctly by changing endstop positions etc. The front left corner of your bed is supposed to be 0,0. 

This tutorial is geared toward voron printers, but applies to any printer. You have to adjust values in your config so when the nozzle is at 0,0 it's over the front left corner of the bed. 

https://docs.vorondesign.com/build/startup/#define-00-point-v0-v1-trident-v2-legacy

Like I said, KAMP doesnt do anything with offsets, it just shrinks the mesh. If the points aren't probed correctly, that is because your machine isn't quite configured correctly.

AxMod3DPrint, 2023-04-17:
The endstop positions are correct. When the printer is told to go to X0 Y0 it does. 

```
[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
rotation_distance: 40
microsteps: 32
endstop_pin: ^!ar3
position_endstop = -21.5
position_min = -21.5
position_max = 220
homing_speed = 60

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
rotation_distance: 40
microsteps: 32
endstop_pin = ^ar14
position_endstop = -6
position_min = -6
position_max = 220
homing_speed = 60
```

The probing positions are relative to the Nozzle, so therefore if I put -40.3 (- being to the right of the nozzle, + being to the left) as my probe X start, the nozzle is at X0 but the probe is 40.3mm in to the bed. I have to put the -61.7 to get the probe where it needs to be. 

Something is funky somewhere, I just can't narrow it down to this script or my config.

kyleisah, 2023-04-17:
I think you might have those offsets flipped, the probe being to the right of the nozzle would be a positive offset, left would be a negative offset. At least, I'm pretty sure. It's been a long time since I've used offsets.

kyleisah, 2023-04-17:
Which would actually kind of make sense because your nozzle is where your probe *should* be, so I think you've got them flipped. 😅

AxMod3DPrint, 2023-04-18:
Yup, that's done it. The things you find when you try and make a mod...

closed 2023-04-18 by AxMod3DPrint

kyleisah, 2023-04-18:
Heh, no worries! Glad it's worked out now! Happy KAMPing! 😄