kyleisah/Klipper-Adaptive-Meshing-Purging #56: Travel speed to the line purge is too fast

cy1der, 2023-04-16:
[image]
I noticed here the travel speed is set to 60 times the max velocity?? Is this intentional?

https://user-images.githubusercontent.com/67352978/232336388-40f6a626-41a1-40d0-9790-b9b1f67095de.mp4
Apologies for above video being distorted.
As you can see/hear it sounds like the belt is going to experience wear much quicker when the travel speed is this high.

I propose for the addition to be able to set the speed of this move.

kyleisah, 2023-04-16:
The printer stores it's maximum velocity as mm/s, but feedrate is called as mm/m, so you have to multiply it by 60 for the speed value to translate correctly from the printer's config file to usable gcode. Whatever you have under [printer] for max_velocity will be the maximum speed your printer can move. You can adjust that lower if you wish for your printer to not move that fast.

phoenixfire425, 2023-04-19:
Same issue here.  even at a conservative 500 max_velocity.  it moves to fast.
500 * 60 = 30000
That is to fast. 

My printer I have the max_velocity at 2700.   2700*500=162000  That would be an insane feedrate.

kyleisah, 2023-04-19:
What I'm saying is, if either of you ( @cy1der or @phoenixfire425 ) feel this travel move is too fast, it is because you have configured your machine to move faster than you are comfortable with. 

The best (and sometimes worst) part of machines is they do exactly what you tell them to. 

You can lower your max_velocity in your printer.cfg file to the speed you are comfortable with. Stock configurations usually have this value set to 150-200, so the machine will only move as fast as 150-200mm/s.

cy1der, 2023-04-19:
I set it to 300 and it is better now, thank you @kyleisah for the clarification!

closed 2023-04-19 by cy1der

kyleisah, 2023-04-19:
> I set it to 300 and it is better now, thank you @kyleisah for the clarification!

Glad to hear it's better, and I'm happy to clarify! Klipper is a pretty complicated firmware so I enjoy helping others understand it better when I can.