kyleisah/Klipper-Adaptive-Meshing-Purging #57: Purge Line off of Bed

LadPack, 2023-04-17:
Hello,

When printing large objects that are either close to the size of the bed, or objects that are diagonally oriented, sometimes the purge line will be printed in the -y, spilling the purge off the side of the bed..  Is there a min/max purge area?

LadPack, 2023-04-17:
It only occurs on my Y axis, where the minimum is -8 in my printer.cfg, is there something I should change there?

kyleisah, 2023-04-17:
Position_max must be increased or decreased depending on where the nozzle is in relation to 0,0. 

There's a helpful guide here that goes over defining the origin of your bed (0,0) accurately. 

https://docs.vorondesign.com/build/startup/#define-00-point-v0-v1-trident-v2-legacy

kyleisah, 2023-04-17:
This is also a very good guide here:

https://github.com/rootiest/zippy-klipper_config/blob/master/guides/GUIDE-axis_limits.md

You pretty much have to tell klipper what your correct axis limits are.

kyleisah, 2023-04-17:
If that doesn't fix it for you, then you will have to wait until I push the next update, as this is currently being worked on in my dev branch.

LadPack, 2023-04-17:
Thanks! I did a similar guide a while back but must have set something wrong.  I'll give it another go

kyleisah, 2023-04-20:
@LadPack just wanted to give you an update, adding a purge_margin variable to KAMP in the next release. I've also worked out how to make it move the purge to the side of the bed if the front of the bed has no room. 😄 

[image]

LadPack, 2023-04-20:
That is awesome! Thank you for the follow up as well, keep up the good work!

closed 2023-07-06 by kyleisah