kyleisah/Klipper-Adaptive-Meshing-Purging #59: Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

cmerkle, 2023-04-18:
When attempting to use adaptive meshing for the first time, I recieved the following error:

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

I followed all the setup instructions, and exclude objects is setup and working. My start print macro is shown below:

```
[gcode_macro START_PRINT]
gcode:
    OFFICE_LIGHTS_ON
    {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(200)|float %}

    STATUS_HEATING
    M104 S{EXTRUDER_TEMP}
    M140 S{BED_TEMP}
    SET_GCODE_OFFSET Z=0.0
    G92 E0 ; Reset Extruder
    HOME_IF_NEEDED
    STATUS_HEATING
    M109 S{EXTRUDER_TEMP} ; Wait for hotend temperature
    M190 S{BED_TEMP} ; Wait for bed temperature
    SETUP_KAMP_MESHING
    BED_MESH_CLEAR
    BED_MESH_CALIBRATE
    STATUS_PRINTING
    G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
    SETUP_LINE_PURGE FLOW_RATE = 4
    LINE_PURGE
```

kyleisah, 2023-04-18:
If there's no polygon attribute, then exclude_object information is not being passed to KAMP. What slicer are you using?

cmerkle, 2023-04-18:
I actually just found more info on the issue here: https://github.com/Ultimaker/Cura/issues/14679

It appears to be a bug with Cura 5.3 that causes it to write "NOMESH" instead of "NONMESH". I was able to resolve it using a search and replace postprocessing script in Cura.

kyleisah, 2023-04-18:
Ah yeah, that's exactly what I was thinking was happening. Glad you got it worked out! Happy KAMPing! 😄

closed 2023-04-18 by kyleisah