kyleisah/Klipper-Adaptive-Meshing-Purging #60: Adaptive Mesh not working

copperfield74, 2023-04-19:
Hi, today I tried to implement adaptive mesh but it always does a complete mesh probing instead of the size of the print object.
My system is updated to:
klipper
v0.11.0-175-gcba119db 
moonraker
v0.8.0-35-geff0901 
mainsail
v2.5.1

Within my printer.cfg I've inserted:
[include Adaptive_Mesh.cfg]
[exclude_object]

in Moonraker.cfg:
[file_manager]
# config_path: ~/klipper_config
# log_path: ~/klipper_logs
# post processing for object cancel. Not recommended for low resource SBCs such as a Pi Zero. Default False 
enable_object_processing: True

my START_PRINT looks like this:
[gcode_macro START_PRINT]
# Reference https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#gcode_macroA
# On how to override default parameters

gcode:
    {% set BED_TEMP = params.BED_TEMP|default(65)|float %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(200)|float %}
    # Start bed heating
    M140 S{BED_TEMP}
    # Use absolute coordinates
    G90
    # Reset the G-Code Z offset (adjust Z offset if needed)
    # SET_GCODE_OFFSET Z=0.0
    # Home the printer
    G28
    BED_MESH_CALIBRATE ; run Bed Mesh
    G1 Z3 F3000 					; move z up little to prevent scratching of surface
    G1 X20 Y5 Z0.3 F5000.0				; move to start-line position
    G1 Z0.3 F1000 				; print height
    G1 X200 Y5 F1500.0 E15 				; draw 1st line
    G1 X200 Y5.3 Z0.3 F5000.0				; move to side a little
    G1 X5.3  Y5.3 Z0.3 F1500.0 E30 			; draw 2nd line
    G1 Z3 F3000 					; move z up little to prevent scratching of surface
    G92 E0 					; Reset Extruder
    # Wait for bed to reach temperature
    M190 S{BED_TEMP}
    # Set and wait for nozzle to reach temperature
    M109 S{EXTRUDER_TEMP}

and in my Ideamaker settings I've inserted the M117 before the START_PRINT
[image]

The console output is:
13:02
G91
G1 Z+25 F1500
G90
13:02
Bed Mesh state has been saved to profile [default]
for the current session. The SAVE_CONFIG command will
update the printer config file and restart the printer.
13:02
Mesh Bed Leveling Complete
13:02
probe at 251.200,299.700 is z=1.738566
13:02
probe at 251.200,299.700 is z=1.743609
13:02
probe at 189.200,299.700 is z=1.695694
13:02
probe at 189.200,299.700 is z=1.693172
13:02
probe at 127.200,299.700 is z=1.660387
13:02
probe at 127.200,299.700 is z=1.657866
13:02
probe at 65.200,299.700 is z=1.640212
13:02
probe at 65.200,299.700 is z=1.642734
13:02
probe at 3.200,299.700 is z=1.627603
13:02
probe at 3.200,299.700 is z=1.625081
13:01
probe at 3.200,242.700 is z=1.718391
13:01
probe at 3.200,242.700 is z=1.718391
13:01
probe at 65.200,242.700 is z=1.688128
13:01
probe at 65.200,242.700 is z=1.683084
13:01
probe at 127.200,242.700 is z=1.690650
13:01
probe at 127.200,242.700 is z=1.698216
13:01
probe at 189.200,242.700 is z=1.662909
13:01
probe at 189.200,242.700 is z=1.665431
13:01
probe at 251.200,242.700 is z=1.672997
13:01
probe at 251.200,242.700 is z=1.678041
13:01
probe at 251.200,185.700 is z=1.718391
13:01
probe at 251.200,185.700 is z=1.720912
13:01
probe at 189.200,185.700 is z=1.746131
13:01
probe at 189.200,185.700 is z=1.753697
13:01
probe at 127.200,185.700 is z=1.758741
13:01
probe at 127.200,185.700 is z=1.758741
13:01
probe at 65.200,185.700 is z=1.731000
13:01
probe at 65.200,185.700 is z=1.728478
13:01
probe at 3.200,185.700 is z=1.786481
13:01
probe at 3.200,185.700 is z=1.783959
13:01
probe at 3.200,128.700 is z=1.746131
13:01
probe at 3.200,128.700 is z=1.743609
13:01
probe at 65.200,128.700 is z=1.736044
13:01
probe at 65.200,128.700 is z=1.736044
13:01
probe at 127.200,128.700 is z=1.761262
13:01
probe at 127.200,128.700 is z=1.761262
13:00
probe at 189.200,128.700 is z=1.763784
13:00
probe at 189.200,128.700 is z=1.766306
13:00
probe at 251.200,128.700 is z=1.743609
13:00
probe at 251.200,128.700 is z=1.743609
13:00
probe at 251.200,71.700 is z=1.778916
13:00
probe at 251.200,71.700 is z=1.781437
13:00
probe at 189.200,71.700 is z=1.814222
13:00
probe at 189.200,71.700 is z=1.809178
13:00
probe at 127.200,71.700 is z=1.786481
13:00
probe at 127.200,71.700 is z=1.789003
13:00
probe at 65.200,71.700 is z=1.723434
13:00
probe at 65.200,71.700 is z=1.720912
13:00
probe at 3.200,71.700 is z=1.751175
13:00
probe at 3.200,71.700 is z=1.748653
13:00
24 | (251.2, 299.7) | (290.0, 258.0)
13:00
23 | (189.2, 299.7) | (228.0, 258.0)
13:00
22 | (127.2, 299.7) | (166.0, 258.0)
13:00
21 | (65.2, 299.7) | (104.0, 258.0)
13:00
20 | (3.2, 299.7) | (42.0, 258.0)
13:00
19 | (3.2, 242.7) | (42.0, 201.0)
13:00
18 | (65.2, 242.7) | (104.0, 201.0)
13:00
17 | (127.2, 242.7) | (166.0, 201.0)
13:00
16 | (189.2, 242.7) | (228.0, 201.0)
13:00
15 | (251.2, 242.7) | (290.0, 201.0)
13:00
14 | (251.2, 185.7) | (290.0, 144.0)
13:00
13 | (189.2, 185.7) | (228.0, 144.0)
13:00
12 | (127.2, 185.7) | (166.0, 144.0)
13:00
11 | (65.2, 185.7) | (104.0, 144.0)
13:00
10 | (3.2, 185.7) | (42.0, 144.0)
13:00
9 | (3.2, 128.7) | (42.0, 87.0)
13:00
8 | (65.2, 128.7) | (104.0, 87.0)
13:00
7 | (127.2, 128.7) | (166.0, 87.0)
13:00
6 | (189.2, 128.7) | (228.0, 87.0)
13:00
5 | (251.2, 128.7) | (290.0, 87.0)
13:00
4 | (251.2, 71.7) | (290.0, 30.0)
13:00
3 | (189.2, 71.7) | (228.0, 30.0)
13:00
2 | (127.2, 71.7) | (166.0, 30.0)
13:00
1 | (65.2, 71.7) | (104.0, 30.0)
13:00
0 | (3.2, 71.7) | (42.0, 30.0)
13:00
bed_mesh: generated points
Index | Tool Adjusted | Probe
13:00
Generating new points...
13:00
Points: x: 5, y: 5
13:00
Algorithm: lagrange
13:00
Object bounds, clamped to the bed_mesh: (42.0, 30.0), (290.0, 258.0)
13:00
0 object points, clamping to bed mesh [(42.0, 30.0) (290.0, 258.0)]
13:00
detach_macro: 'Dock_Probe'
13:00
attach_macro: 'Attach_Probe'
13:00
probe_dock_enable: 0
13:00
fuzz_max : 4.000000
13:00
fuzz_min : 0.000000
13:00
fuzz_enable : 0
13:00
status_macro: 'status_meshing'
13:00
led_enable : 0
13:00
Unknown command:"T0"
12:57
File selected
12:57
File opened:Flow_Cube_30.gcode Size:477811

I've already restartet everything, but still no adaptive mesh. I use a BLTouch for probing on my Sidewinder X1.

Thanks for your help.

kyleisah, 2023-04-19:
`0 object points, clamping to bed mesh [(42.0, 30.0) (290.0, 258.0)]`

This is your problem right here. There is no exclude_object information being fed to KAMP.

What slicer are you using?

kyleisah, 2023-04-19:
Oh I see now, looks like ideamaker. Does that slicer support labeling objects? That is what KAMP needs to work any of it's magic.

copperfield74, 2023-04-19:
How can I check the gcode for labelling?

copperfield74, 2023-04-19:
[Flow_Cube2x_30.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11273995/Flow_Cube2x_30.txt)

Here is a gcode that has two objects on the plate.

kyleisah, 2023-04-19:
```gcode
;Bounding Box: 131.600 202.400 131.600 168.400 0.000 30.160
;Model Gap: 0.000
M221 T0 S96.00
M140 S75.00
M104 T0 S210.00
M109 T0 S210.00
T0
M190 S75.00
M117
START_PRINT EXTRUDER_TEMP=210 BED_TEMP=75
;TOTAL_NUM: 2
;PRINTING_TIME: 0
;REMAINING_TIME: 2839
M83
;LAYER:0
;Z:0.400
;HEIGHT:0.400
M106 S115
M104 T0 S215
;PRINTING: NON-OBJECT
;PRINTING_ID: -1
G1 F2700 E-0.8000
M204 S1500.00
M205 X7.00 Y7.00
G0 F7200 X132.799 Y132.411
G0 F900 Z0.400
M204 S800.00
```

Hm, yeah it looks like EXCLUDE_OBJECT_DEFINE is missing. That is usually added by the object cancellation preprocessor when the gcode is uploaded. So, it looks like the preprocessor didn't run. 

I've seen this before, only once, but I saw it during early dev. I had to reinstall moonraker and that fixed it. 

Did you by any chance install klipper, moonraker, mainsail, etc with KIAUH?

kyleisah, 2023-04-19:
You can try running these commands to reinstall moonraker, then you will need to reupload the gcode from your slicer, and check the gcode for an EXCLUDE_OBJECT_DEFINE macro somewhere near the top. 

```ini
cd ~/moonraker
git pull
./scripts/install-moonraker.sh -r
```

Let me know if that works!

copperfield74, 2023-04-19:
I'm sorry, I didn't know that the files are altered as soon you upload them.
The first file was directly from Ideamaker.
This file here is downloaded from Klipper and has the EXCLUDE options.
[Flow_Cube2x_30_new.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11274232/Flow_Cube2x_30_new.txt)

kyleisah, 2023-04-19:
@copperfield74 ah there it is! Yes, this is typically how that works. The gcode is sent to the machine, the machine processes the object(s) and then shows it in the UI. 

Since it's showing that now, KAMP should have no problem working. 😄

kyleisah, 2023-04-19:
Oh wait, it's showing the exclude_object macros *after* print_start is called. That's not right. Hmmm. Let me see what else we can try.

kyleisah, 2023-04-19:
Hmm, nope, you've got M117 before Print_Start, so it should have put it in the right spot to be loaded correctly. That's odd. 

Remove those gcode files from your mainsail instance, and re-slice and upload them. exclude_object (currently) processes objects after the first command it sees. Unless @kageurufu has made any updates, but I don't know for sure if that's happened yet.

kyleisah, 2023-04-19:
@copperfield74 I'm sorry you're having these issues, this isn't typical. I'm interested if reinstalling moonraker would fix this, could you try that for me?

copperfield74, 2023-04-19:
Your help is highly appreciated. I've reinstalled moonraker and uploaded my file once again but the structure is still the same as before. :-(

CodeMonkeyX, 2023-04-21:
Just an addition I also seem to be having this issue. I have two printers one with Mainsail and one with Fluidd (although I do not think that should have any effect on this?) My Fluidd V2.4 printer works with adaptive mesh just fine. My Mainsail Trident printer does not. 

The Trident printer is doing the same thing as described by @copperfield74, the gcode file does not seem to be getting processed by Moonraker when it's imported. I tried importing the gcode file to both machines, one adds the exclude objects and the other does not.

I am going to keep messing around with stuff and will post if I find anything that makes it work. :P

CodeMonkeyX, 2023-04-21:
Ok I fixed mine. For some reason my Moonraker install was looking for it's config file in the wrong place. It was looking in `/home/pi/klipper_config/moonraker.conf` and I was editing `/home/pi/printer_data/config/moonraker.conf` when I was trying to enable object processing. 

If you check you Moonraker log file search for `config_file:' and make sure it matches the location of the file you are editing.

Amazingly enough I was not sure how to change this, so I asked ChatGPT and it actually told me how...
>To change the default location of the moonraker.conf file, you can use the -c command-line option when running the install >script (install-moonraker.sh). [For example, you can run ./install-moonraker.sh -c /path/to/moonraker.conf to specify a custom >path for the configuration file](https://moonraker.readthedocs.io/en/latest/user_changes/)[1](https://moonraker.readthedocs.io>>/en/latest/user_changes/). [Alternatively, you can edit the environment file located at >~/printer_data/systemd/moonraker.env and change the value of MOONRAKER_CONFIG_PATH to point to your desired location]>(https://github.com/Arksine/moonraker/blob/master/docs/user_changes.md)[2](https://github.com/Arksine/moonraker>/blob/master/docs/user_changes.md).

I did the 2nd thing, I modified the `moonraker.env` to point to `-c /home/pi/printer_data/config/moonraker.conf` restarted Moonraker, and now it all works correctly.

Hopefully the next time I update Moonraker it does not change it back? But at least I know to check that now.

og54, 2023-04-25:
Adaptive meshing does not seem to work when exporting and auto starting prints via the Moonraker plugins in Superslicer or Cura. It works when exporting and manually starting the print.
Adding the header to the resulting gcode below:
`; external perimeters extrusion width = 0.60mm
; perimeters extrusion width = 0.69mm
; infill extrusion width = 0.96mm
; solid infill extrusion width = 0.84mm
; top infill extrusion width = 0.60mm
; first layer extrusion width = 0.75mm

; object:{"name":"[a]_latch_shuttle","id":"[a]_latch_shuttle.stl id:0 copy 0","object_center":[169.306010,181.666354,0.000000],"boundingbox_center":[169.306010,181.666354,5.200001],"boundingbox_size":[22.966833,17.888338,10.400002]}
; object:{"name":"[a]_latch_shuttle","id":"[a]_latch_shuttle.stl id:0 copy 1","object_center":[180.693879,168.333901,0.000000],"boundingbox_center":[180.693879,168.333901,5.200001],"boundingbox_size":[22.966833,17.888338,10.400002]}
; plater:{"center":[174.999944,175.000127,0.000000],"boundingbox_center":[174.999944,175.000127,5.200001],"boundingbox_size":[34.354702,31.220791,10.400002]}

;TYPE:Custom
; M190 S0
; M109 S0 ; uncomment to remove set&wait temp gcode added automatically after this start gcode
PRINT_START TEMP_EXTRUDER=255 TEMP_BED=110 TEMP_CHAMBER=50

; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 2 known objects
EXCLUDE_OBJECT_DEFINE NAME=a__latch_shuttle_stl_id_0_copy_0 CENTER=167.682,179.492 POLYGON=[[158.127,173.024],[158.127,190.308],[180.488,190.308],[180.488,173.024]]
EXCLUDE_OBJECT_DEFINE NAME=a__latch_shuttle_stl_id_0_copy_1 CENTER=179.068,166.161 POLYGON=[[169.515,159.691],[169.515,176.976],[191.876,176.976],[191.876,159.691]]
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion`

maz0r, 2023-04-27:
> Adaptive meshing does not seem to work when exporting and auto starting prints via the Moonraker plugins in Superslicer or Cura.

This is likely due to the plugin queuing commands and preventing moonraker's object-postprocessor being able to run before the start is triggered. 

You could move the post-processor into your slicers workflow per https://github.com/kageurufu/preprocess_cancellation#superslicer-prusaslicer-and-slic3r this should ensure the object code is amended before the code is uploaded to the printer.

closed 2023-07-06 by kyleisah