kyleisah/Klipper-Adaptive-Meshing-Purging #62: Using with bltouch

gazialankus, 2023-04-21:
Since bltouch is not mentioned anywhere, I thought I'd ask to be sure. Which of the three setup options listed here makes sense ofr a bltouch owner? https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#setup 

Thanks in advance.

david1111967, 2023-04-23:
Make the config normally validate that the variable_probe_dock_enable line is false.
And this should work

kyleisah, 2023-04-23:
@gazialankus because the BLTouch is attached to the printhead and is not dockable, it is considered integrated.

closed 2023-04-26 by kyleisah

gazialankus, 2023-04-26:
Ok so it's this one, thank you for clarification! https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#ive-configured-my-probe-already-or-my-probe-is-integrated