kyleisah/Klipper-Adaptive-Meshing-Purging #65: Move exceeds maximum extrusion (4.165mm^2 vs 0.640mm^2)

dgpmaluco, 2023-04-22:
Hello guys,

Not sure what I'm missing here.
Trying to set up the voron purge and I get this error after the Adaptive Mesh.

What could be the problem?
Tried lowering the "variable_flow_rate: 10" to 1.5 and that didn't help

CodeMonkeyX, 2023-04-23:
Look up `max_extrude_cross_section:` in the Klipper documentation.

https://www.klipper3d.org/Config_Reference.html#extruder

The purge macro calls for a lot more plastic to be extruded than is normal. I had to increase my value to just a little higher than what the error was telling me. Then it should work.

If you have it set to 0.640mm^2 then that is pretty low. It should be around

dgpmaluco, 2023-04-23:
Yep! That solved it... Thank you very much :)

closed 2023-04-26 by kyleisah