kyleisah/Klipper-Adaptive-Meshing-Purging #66: Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'POLYGON'

horatio1989, 2023-04-24:
When trying to use adaptive mesh on klipper and Mainsail (Using prusaslicer), I got the error:

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'POLYGON'

I have checked the gcode after was uploaded, and the exclude object is working and adds at the begining:

EXCLUDE_OBJECT_DEFINE NAME=Part_Studio_1_Part_2_stl_id_0_copy_0 CENTER=137.618,101.540 POLYGON=[[125.347,89.282],[125.347,113.832],[149.897,113.832],[149.897,89.282]]
EXCLUDE_OBJECT_DEFINE NAME=Part_Studio_2_stl_id_1_copy_0 CENTER=99.401,102.546 POLYGON=[[87.129,90.288],[87.129,114.838],[111.679,114.838],[111.679,90.288]]
;TYPE:Custom

and then the rest of the file is ok.

I have no idea why is it not working, since there is no other instance in the gcode on which appears "EXCLUDE_OBJECT_DEFINE NAME" without "POLYGON"

On each gcode block for each object you see the end and start:

; stop printing object Part Studio 1 - Part 2.stl id:0 copy 0
EXCLUDE_OBJECT_END NAME=Part_Studio_1_Part_2_stl_id_0_copy_0
; printing object Part Studio 2.stl id:1 copy 0
EXCLUDE_OBJECT_START NAME=Part_Studio_2_stl_id_1_copy_0

Has anyone any idea what could be the problem?

DasLukas, 2023-04-24:
Did you add `[exclude_object]` to your printer.cfg. [https://www.klipper3d.org/Config_Reference.html#exclude_object](https://www.klipper3d.org/Config_Reference.html#exclude_object)

horatio1989, 2023-04-24:
Yes! Sorry I didn’t explain. I already have the exclude object working properly in mainsail with prusa slicer. The adaptive mesh is what is not working.

My start print macro is just the line to bed_mesh_calibration

DasLukas, 2023-04-24:
Okay. If I understood you right, correct me if I'm wrong, then your `START_PRINT` macro only contains `BED_MESH_CALIBRATION`?

Can you poste your `START_PRINT` gcode macro and maybe upload your `klippy.log`. More input should be nice.

horatio1989, 2023-04-25:
Yes, you understood right! here is my START_PRINT macro:

`[gcode_macro START_PRINT]
gcode:
    M117 Debugging
    BED_MESH_CALIBRATE`

and the [klippy.log](https://github.com/horatio1989/Klipper_Mr.H/blob/main/klippy.log)

DasLukas, 2023-04-25:
First of all I mention you follow the steps described in the [setup](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#ive-configured-my-probe-already-or-my-probe-is-integrated) section of this repository. So I mention you add the following to your `moonraker.conf`:

```
[file_manager]
enable_object_processing: True
...
```

What I see in your klippy.log, respectively in your `START_PRINT` macro, is that you don't run the setup macros before the `BED_MESH_CALIBRATE` macro. But didn't run the setup macros should not cause this error. It only runs the non adaptiv bed mesh.

Here a example `START_PRINT` macro:

```
[gcode_macro START_PRINT]
gcode:
    [...]

    SETUP_KAMP_MESHING
    SETUP_LINE_PURGE

    BED_MESH_CLEAR
    BED_MESH_CALIBRATE
    LINE_PURGE

    [...]
```

But what cause your Problem? Did you see this troubleshooting guide [https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#troubleshooting](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#troubleshooting)? I see you add a M117 to your `START_PRINT` macro but maybe you try to add it before calling `START_PRINT` in your slicers start gcode section.

horatio1989, 2023-04-25:
Yes, I follow those steps! I have enable_object_processing set to true in moonraker.conf

I tried adding the macros that you suggested. same issue

I have look up the troubleshooting, and I have add the M117 before the macro, and inside the macro, with no luck., same issue

Necroknight2507, 2023-04-25:
I have the same error usig cura.
Cura inserts a strange line at the start of the gcode file:

EXCLUDE_OBJECT_DEFINE NAME=NOMESH

deleting it "fixes" the error

ohrn, 2023-05-06:
The line

EXCLUDE_OBJECT_DEFINE NAME=NOMESH

is added by the Moonraker object processor when handling files sliced by Cura. Normally this is supports but apparently it always gets emitted by Cura even if empty.

My Jinja isn't the best but I worked around it by changing the line

{% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}

 to

{% set all_points = printer.exclude_object.objects | selectattr('polygon') | map(attribute='polygon') | sum(start=[]) %}

In the KAMP macros.

ohrn, 2023-05-06:
Even easier, turns out upgrading Cura to 5.3.1 makes the NOMESH problem go away.

closed 2023-07-06 by kyleisah