kyleisah/Klipper-Adaptive-Meshing-Purging #67: KAMP on my v2.4 not working right

rmilyard, 2023-04-25:
I setup a new v2.4 350.  When bed_mesh_calibrate runs still wants to do the whole bed not just where print will be.  I have all the setting in slicer and printer.cfg setup right I think.

DasLukas, 2023-04-25:
I guess you follow all steps described on the main page of this repository. If I understood your right, you want to run in some cases a non adaptiv bed mesh and in some other cases a normal bed mesh for the whole bed.

Some times I have the same approaches. To achieve this I use a START_PRINT macro where I can switch adaptiv on or off.

```
[gcode_macro START_PRINT]
gcode:
    {% set pa = params.PA|default(0.0)|float %}

    [...]

    G28 ; Home the printer

    {% if adaptiv >= 1%}
      SETUP_KAMP_MESHING
      SETUP_LINE_PURGE
    {% endif %}

    BED_MESH_CLEAR
    BED_MESH_CALIBRATE

    [...]

    {% if adaptiv == 1 %}
      LINE_PURGE
    {% else %}
      [...]
    {% endif %}
```

In your slicers start gcode section run `START_PRINT ADAPTIV=0` for non adaptiv bed mesh.

If you want to use KAMP and always probe you whole mesh, I not see any advantages.

Is your problem not solved, please provide your slicers start gcode and your klippy.log.

kyleisah, 2023-04-26:
@rmilyard 

I'm going to need more information before I can help. 

Send a gcode file and a klipper log please

closed 2023-07-06 by kyleisah