kyleisah/Klipper-Adaptive-Meshing-Purging #68: Adaptive Purging tries to go off bed

KittenRaee, 2023-04-25:
Both Line Purge and Voron Purge can and will attempt to crash print head at maximum possible speed making their best to go past physical bounds of a printer. This is prevalent on Voron v0, where every millimetre counts tenfold and printing anything bigger (eg. StealthBurner main cowling) will put adaptive purges well past bounds of the tiny printer.

So far the only solution is to use an exclusion region in slicer and a static purge position.

kyleisah, 2023-04-26:
It doesn't do that on my V0. I suggest you move your bed (if you have newer kirigami with slotted holes) or adjust the maximums you have defined for each axis.

closed 2023-04-26 by kyleisah

kyleisah, 2023-04-26:
Also, there are provisions being added to the line_purge macro to purge along the side of the bed if there's no room in the front.