kyleisah/Klipper-Adaptive-Meshing-Purging #69: first layer problems - mesh loaded but not applied (maybe)

FkYeahScience, 2023-04-25:
Hey everyone

got it to work successfully on my Voron v0.1 Mainsail with sideswipe deployable probe and sensorless in XY.
https://github.com/oldfar-t/Side-Swipe-Magnetic-Probe/

My bed is pretty crooked for its size, BED_MESH_CALIBRATE gives +-0.5mm at 110° for 60+ mins; after successful BED_CREWS_ADJUST 

I am using the z-probe as a virtual z endstop.

Z homing works (the initial Z homing happens at X60 Y60 (nozzle, not probe) as defined in the [homing_override] -  BUT without the offset calculated in (so the probe measures Z0 at X59 Y60-28), UI showing X60 Y60 whilst measuring in Z)

probe_accuracy range 0.008 (omron D2F)
probe XY offset is correct (boundaries are used correctly with BED_MESH_CALIBRATE)
Probe Z offset seems correct for a test cube 20^3
mesh is correctly loaded (in mesh visualisation)
filament accuracy ist < +-0,02

when using KAMP on a part tho - the meshing seems to use the right xy coordinates, they all have the same x/y offset from the part, when i print bigger parts i do not have a flat first layer (partly perfect)

is there another offset I have to change? 

printer.cfg
[include mainsail.cfg]
[include adaptive_mesh.cfg]
[include voron_purge.cfg]
[exclude_object]
[…]

[stepper_z]
endstop_pin: probe: z_virtual_endstop
[…]

[bed_mesh]
relative_reference_index: 0
[…]

[probe]
z_offset: 2.64 
x_offset: -1
y_offset: -28
[…]

PROBE_OUT                      #attach probe to toolhead
PROBE_IN                          #detach probe from toolhead
[…]

________________________________________

moonraker.cfg
enable_object_processing: True
[…]

________________________________________

adaptive_mesh.cfg
variable_probe_dock_enable: True
variable_attach_macro: ‘PROBE_OUT‘
variable_detach_macro: ‘PROBE_IN‘

________________________________________

SuperSlicer:
Label objects: True

________________________________________

autoupdating KAMP was not enabled

FkYeahScience, 2023-04-25:
here are my configs, for full reference

[adaptive_mesh.cfg.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11323275/adaptive_mesh.cfg.txt)
[mainsail.cfg.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11323277/mainsail.cfg.txt)
[printer.cfg.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11323278/printer.cfg.txt)
[voron_purge.cfg.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11323279/voron_purge.cfg.txt)

kyleisah, 2023-04-26:
All KAMP does is adjust the bounds of your bed mesh, nothing more. If you're having inconsistent first layers, you likely have a mechanical issue.

closed 2023-04-26 by kyleisah