kyleisah/Klipper-Adaptive-Meshing-Purging #73: cant get it to work

badplay, 2023-04-26:
becorse body is too long i added files
gcode and printer.cfg
[printercfg.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11335530/printercfg.txt)
[gcodespacer.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11335531/gcodespacer.txt)

were is a problem??? 

and is there a way to use  normal bad mesh mode /  adaptiv  inside klipper?

blazecoding2009, 2023-04-26:
What's the problem?

kyleisah, 2023-04-27:
I need more information than "can't get it to work".

badplay, 2023-04-27:
i have edit my gcode to BED_MESH_CALIBRATE  put out the BED_MESH_CALIBRATE Load: default

now i got 

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

is there a way to use full bed mesh  if needed so is there a way to make switchable from full to kamp ??

maz0r, 2023-04-27:
@badplay I can see you are using `Generated with Cura_SteamEngine 5.3.0` to slice your file, this version of Cura has a known bug, and a fix was merged on March 24th per https://github.com/Ultimaker/Cura/issues/14679 and is included with Cura 5.3.1.

DasLukas, 2023-04-27:
To make the adaptiv mesh optional I use a `START_PRINT` macro that I already published [here](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/issues/67#issuecomment-1521526066).

kyleisah, 2023-04-27:
@DasLukas i appreciate the help, but please do not direct users who have issues to use unsupported installations/configuration. 

If you want to toggle adaptive mesh on and off, simple comment out `[include Adaptive_Mesh.cfg]`.

kyleisah, 2023-05-03:
Glad it's working!

Any moonraker update info is all found in the readme. It's not necessary at all. Some just prefer it.

closed 2023-05-03 by kyleisah