kyleisah/Klipper-Adaptive-Meshing-Purging #74: KAMP File in mainsail etc. showing up as (read only)

DoomedPaints, 2023-04-27:
**Describe the issue**
A clear and concise description of what the bug is that the .cfg files in the KAMP folder are coming in as (read only) whis makes them non-executable

**What slicer are you using?**
SuperSlicer

**Troubleshooting**
Deleting all files from the folder, creating new file and copy/pasting from raw contents on github repo

kyleisah, 2023-04-27:
That's really odd. Does your user (`pi`) have write access to those files?

kyleisah, 2023-04-27:
You'd have to `ssh` into your machine, `cd` into where your config files are located, and use the command `ls -l` to list files and their permissions

thelittlemike, 2023-04-29:
Same issue here. Followed the instructions by git pulling to the home directory and then creating symbolic link. If I ssh into the Pi, I can edit the files fine. But if I try to open them through Mainsail, they all open read-only. Likely has something to do with how Mainsail changed some things around in October.

JEDean, 2023-04-29:
Same Issue as well.:

[image]

Gabbajoe, 2023-04-30:
Hey that's normal in mainsail (fluid i guess too) for symbolic linked files (ln -s) in the linux file system. I think this is a conscious choice in the editor of Mainsail (fluid too?). Because if you edit the file in the git repo you linked it from you create a conflict and a simple git pull to update from the remote git repo doesn't work anymore. That's why mainsail has overwrite able variables for its PAUSE/ABORT macro.

Example here: my config directory:
`pi@vtwo:~/printer_data/config $ ls -l
total 2056
-rw-r--r-- 1 pi pi   5822 Jan 14 03:01 Adaptive_Mesh.cfg
drwxr-xr-x 2 pi pi   4096 Feb  2  2022 backup
drwxr-xr-x 2 pi pi   4096 Jan 17 19:00 btt-octopus
lrwxrwxrwx 1 pi pi     55 Apr 22 20:25 KAMP -> /home/pi/Klipper-Adaptive-Meshing-Purging/Configuration
drwxr-xr-x 2 pi pi   4096 Jan 17 19:00 linuxhost
lrwxrwxrwx 1 pi pi     37 Jan 13 23:50 mainsail.cfg -> /home/pi/mainsail-config/mainsail.cfg
lrwxrwxrwx 1 pi pi     55 Jan 13 23:49 mainsail-moonraker-update.conf -> /home/pi/mainsail-config/mainsail-moonraker-update.conf
-rw-r--r-- 1 pi pi    908 Apr 22 20:37 moonraker.conf
-rw-r--r-- 1 pi pi  12384 Oct 15  2021 nozzle_srub.cfg
lrwxrwxrwx 1 pi pi     37 Mar 24 01:23 output -> /home/pi/probe_accuracy_tests/output/
-rw-r--r-- 1 pi pi  16758 Apr 22 22:38 printer.cfg
-rw-r--r-- 1 pi pi   3416 Apr  5  2022 resonance.cfg
-rw-r--r-- 1 pi pi   2279 Jan 20 22:22 runout.cfg
-rw-r--r-- 1 pi pi 166783 Apr 16 18:14 shaper_calibrate_x_20230416_181409.png
-rw-r--r-- 1 pi pi 170672 Apr 16 18:22 shaper_calibrate_x_20230416_182235.png
-rw-r--r-- 1 pi pi 170925 Apr 16 18:35 shaper_calibrate_x_20230416_183452.png
-rw-r--r-- 1 pi pi 180781 Apr 16 18:39 shaper_calibrate_x_20230416_183905.png
-rw-r--r-- 1 pi pi 165518 Apr 16 18:48 shaper_calibrate_x_20230416_184757.png
-rw-r--r-- 1 pi pi 154903 Jan 21 13:05 shaper_calibrate_x.png
-rw-r--r-- 1 pi pi 163840 Apr 16 18:14 shaper_calibrate_y_20230416_181419.png
-rw-r--r-- 1 pi pi 160336 Apr 16 18:22 shaper_calibrate_y_20230416_182245.png
-rw-r--r-- 1 pi pi 158229 Apr 16 18:35 shaper_calibrate_y_20230416_183505.png
-rw-r--r-- 1 pi pi 170339 Apr 16 18:39 shaper_calibrate_y_20230416_183915.png
-rw-r--r-- 1 pi pi 169347 Apr 16 18:48 shaper_calibrate_y_20230416_184808.png
-rw-r--r-- 1 pi pi 161723 Jan 21 13:05 shaper_calibrate_y.png
-rw-r--r-- 1 pi pi  10262 Jan 14 00:03 stealthburner_leds.cfg
-rw-r--r-- 1 pi pi    166 Oct  6  2021 temp.cfg
-rw-r--r-- 1 pi pi   2592 Jan 29  2022 webcam.txt`

In mainsail the files mainsail.cfg, mainsail-moonraker-update.conf and all files in KAMP are read only. 

[image]
[image]

Cheers and just my 2 cents to this issues ;)
Gabbajoe

thelittlemike, 2023-04-30:
> Hey that's normal in mainsail (fluid i guess too) for symbolic linked files (ln -s) in the linux file system. I think this is a conscious choice in the editor of Mainsail (fluid too?). Because if [...]

But I can edit any other symbolically linked files, just not KAMP. So I'm not sure that's the issue. At least in my case, I'm not sure. Never ran into this with anything else.

kyleisah, 2023-04-30:
Please ssh into your pi and try this command to give permissions to the folder:

`chmod 777 ~/Klipper-Adaptive-Meshing-Purging/`

That should give full read and write access to that folder to all users/groups.

thelittlemike, 2023-04-30:
> 

Looks like that worked. I thought I had tried that before, but I guess not.

thelittlemike, 2023-04-30:
I take it back, it didn't work. It removed the "Read-Only" from the top but when I tried to change something, save & restart, it says adaptive_mesh.cfg could not be uploaded (Adaptive Mesh file was what I was trying to edit).

JEDean, 2023-04-30:
I've got the same results as rball8985, though I am able to edit and save the config files via WINSCP directly on the pi. Its access through Fluidd that doesn't want to work. So, at least there is a work around.

JD

kyleisah, 2023-05-03:
I don't know why this is happening, but I think this is some kind of permission issue with your klipper/mainsail installation. 

I'm closing this for now, as there's nothing I can do about that, but if you find a fix, please update me here so I at least have a lead if this happens again

closed 2023-05-03 by kyleisah