kyleisah/Klipper-Adaptive-Meshing-Purging #76: KAMP tries to probe outside of bounds, and I do not know why?

Voxel3DPrinting, 2023-05-01:
**Describe the issue**
I just set up KAMP on my 350mm V2.4R2 after seeing the video of MODbot. I am really happy that the feature exists, however. I cannot get it to work.
Even though I have 1 cube in the middle of the bed, it still probes the whole bed with standard bounds. I have excluded objects defined and labelled objects is on, and I cannot see why it wants to do a complete mesh.

Also, when starting the purge, it gives the error "Move exceeds maximum extrusion (1.924mm^2 vs 0.640mm^2)" What am I going wrong??

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
Moving model around, setting bounds, changing density of mesh 

[Adaptive_Mesh.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11364934/Adaptive_Mesh.zip)

mloebbert, 2023-05-01:
This is caused by the the same issue I reported 10 minutes bevore.

This is the gcode of your print:

;TYPE:Custom
print_start EXTRUDER=215 BED=50

; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 1 known objects
EXCLUDE_OBJECT_DEFINE NAME=Voron_Design_Cube_v7_stl_id_0_copy_0 CENTER=175.465,175.080 POLYGON=[[160.21,160.21],[160.21,189.916],[189.793,189.916],[189.793,160.21]]

 The problem is that 'EXCLUDE_OBJECT_DEFINE' is instered after 'print start'. You can copy the 'EXCLUDE_OBJECT_DEFINE'- line before the 'print start' marco. Than adaptive probing works as expected

Voxel3DPrinting, 2023-05-01:
EDIT;

I just noticed these lines in the console;
Object bounds, clamped to the bed_mesh: (40.0, 40.0), (310.0, 310.0)
0 object points, clamping to bed mesh [(40.0, 40.0) (310.0, 310.0)]

This would indicate that there is something going wrong with defining the bounds of the 3D print.

Also, I figured out that the second error (exceeding max extrusion) is because of a limit. But this might be handy to set in the installation manual

Voxel3DPrinting, 2023-05-01:
> EXCLUDE_OBJECT_DEFINE

ahhh ohkee,

So then, in printer.cfg, you get this start gcode:

```
[gcode_macro PRINT_START]
#   Use PRINT_START for the slicer starting script - please customise for your slicer of choice
gcode:

    {% set BED = params.BED | default(60) | float %}
    {% set EXTRUDER = params.EXTRUDER | default(215) | float %} 
    EXCLUDE_OBJECT_DEFINE
    SAVE_GCODE_STATE NAME=start
    G32                                ; Home XYZ and do QGL
    M117 Starting warmup
    G1 Z20 F3000                       ; Move nozzle away from bed
    M117 Warmup
    M190 S{BED}                        ; Set bed temp and wait for it reach temp
    M109 S{EXTRUDER|int}               ; M109 heat and wait for it to reach temp

    BED_MESH_CALIBRATE                 ; Make mesh of build plate
    BED_MESH_PROFILE LOAD=default      ; Load mesh of build plate

    G1 Z5 F5000                        ; Move head 5mm from bed surface
    M83                                ; Make the E relative independant of other axis
    #M117 Purging
    #PURGE_BUCKET
    #G1 E2 F1500 # unretract
    M117 Starting Print
    LINE_PURGE                         ; Purge hotend
    RESTORE_GCODE_STATE NAME=start
```

Voxel3DPrinting, 2023-05-01:
EDIT:

I added the EXCLUDE_OBJECT_DEFINE line at the front of PRINT_START macro, but it still does **not** work. 

Purging issue was solved by adding "max_extrude_cross_section: 50.0" to the extruder section in printer.cfg

mloebbert, 2023-05-01:
You must add SETUP_KAMP_MESHING and SETUP_LINE_PURGE to your PRINT_START macro before BED_MESHING. This is described in the dokumentation under 'I prefer using Moonraker's update manager'

Voxel3DPrinting, 2023-05-01:
It is still not working...

My startup macro in printer.cfg;
```
[gcode_macro PRINT_START]
#   Use PRINT_START for the slicer starting script - please customise for your slicer of choice
gcode:

    {% set BED = params.BED | default(60) | float %}
    {% set EXTRUDER = params.EXTRUDER | default(215) | float %} 
    SETUP_KAMP_MESHING 
    SETUP_LINE_PURGE 
    SAVE_GCODE_STATE NAME=start
    G32                                ; Home XYZ and do QGL
    M117 Starting warmup
    G1 Z20 F3000                       ; Move nozzle away from bed
    M117 Warmup
    M190 S{BED}                        ; Set bed temp and wait for it reach temp
    M109 S{EXTRUDER|int}               ; M109 heat and wait for it to reach temp

    BED_MESH_CALIBRATE                 ; Make mesh of build plate
    BED_MESH_PROFILE LOAD=default      ; Load mesh of build plate

    G1 Z5 F5000                        ; Move head 5mm from bed surface
    M83                                ; Make the E relative independant of other axis
    #M117 Purging
    #PURGE_BUCKET
    #G1 E2 F1500 # unretract
    M117 Starting Print
    LINE_PURGE                         ; Purge hotend
    RESTORE_GCODE_STATE NAME=start
```

Slicer start gcode script;
```
print_start EXTRUDER=[first_layer_temperature] BED=[first_layer_bed_temperature]

SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
```

[3DBenchy.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11365267/3DBenchy.zip)

Voxel3DPrinting, 2023-05-01:
Fixed it with the M117 line. It is working correctly now.
Thanks for the help!

closed 2023-05-01 by Voxel3DPrinting

mloebbert, 2023-05-01:
great! May you share your sliced working gcode.

Voxel3DPrinting, 2023-05-01:
[3DBenchy (1).zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11365361/3DBenchy.1.zip)

Here you go!

To summarize;
**Issue**;
     - error "Move exceeds maximum extrusion (1.924mm^2 vs 0.640mm^2)"
 **fix**; 
     -  adding "max_extrude_cross_section: 50.0" to the extruder section in printer.cfg
   
    
**Issue**;
     - Not probing adaptive mesh
 **fix**; 
     -  adding "M117" to the start of start gcode in slicer, before start_gcode