kyleisah/Klipper-Adaptive-Meshing-Purging #77: When BED_MESH_CALIBRATION called, error about missing "POLYGON" happens

horatio1989, 2023-05-01:
Every time I sent a print to start, when the `BED_MESH_CALIBRATION` command is sent the terminal gives the following error twice one right after the other:

_Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'POLYGON'_ 

Using:
PrusaSlicer 2.5.2
klipper v0.11.0-190-g19877425
crowsnest v3.0.7-3-g20ed6a8
mainsail-config v1.0.0-11-ge4976f6
moonraker v0.8.0-39-g6c7dfe5
mainsail v2.5.1

I have Exclude Object and `BED_MESH_CALIBRATE` working when not using KAMP. I have followed the instructions on the KAMP github to install it. I have checked the troubleshooting section, but the only thing there is the issue with Cura.

On PrusaSLicer, the Start G-code is the following:
```
M109 S0
M190 S0
M117
START_PRINT BED_TEMP={first_layer_bed_temperature[0]} EXTRUDER_TEMP={first_layer_temperature[0]}
```

My START_PRINT macro is:

`[gcode_macro START_PRINT]
gcode:
    M117 Andiamo!
    SFS_DISABLE                  # Disable smart filament sensor
    G90                          # use absolute coordinates
    M83                          # extruder relative mode
    SKEW_PROFILE LOAD=skew1
    SETUP_KAMP_MESHING DISPLAY_PARAMETERS=1
    SETUP_LINE_PURGE DISPLAY_PARAMETERS=1
    M140 S60 #{% set BED_TEMP = params.BED_TEMP|default(60)|float %}
    M104 S180 #{% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(180)|float %}
    M190 S60
    G28                          # Home all
    Z_TILT_ADJUST			        
    BED_MESH_CLEAR
    BED_MESH_CALIBRATE
    #BED_MESH_PROFILE LOAD=mesh1 # load the Mesh #1
    M104 S{EXTRUDER_TEMP} 	     # set final nozzle temp
    G92 E0					     # set extruder position to 0
    M109 S{EXTRUDER_TEMP} 	     # wait for nozzle temp to stabilize
    LINE_PURGE`

As explained, everything inside the START_PRINT macro works fine up to `BED_MESH_CALIBRATE`

Sliced file, after it was processed by Mainsail:
[Test_3_boxes.gcode.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11365662/Test_3_boxes.gcode.txt)
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11365775/klippy.log)
[moonraker.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11365778/moonraker.log)
[crowsnest.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11365779/crowsnest.log)

I checked on [Klipper website](https://www.klipper3d.org/G-Codes.html#exclude_object_define) about `exclude_object_define` and it looks like the syntax should be:
`EXCLUDE_OBJECT_DEFINE [NAME=object_name [CENTER=X,Y] [POLYGON=[[x,y],...]]`
while my processed gcode looks like:
`EXCLUDE_OBJECT_DEFINE NAME=Shape_Box_id_1_copy_0 CENTER=159.178,101.793 POLYGON=[[147.664,90.307],[147.664,113.357],[170.714,113.357],[170.714,90.307]]`

A couple of brackets are missing. I tried manually adding them, but still same issue. not matter which combination I was using:
option 1: `EXCLUDE_OBJECT_DEFINE NAME=Shape_Box_id_1_copy_0 CENTER=159.178,101.793 [POLYGON=[[147.664,90.307],[147.664,113.357],[170.714,113.357],[170.714,90.307]]]`
option 2: `EXCLUDE_OBJECT_DEFINE NAME=Shape_Box_id_1_copy_0 CENTER=[159.178,101.793] [POLYGON=[[147.664,90.307],[147.664,113.357],[170.714,113.357],[170.714,90.307]]]`
option 3: `EXCLUDE_OBJECT_DEFINE NAME=Shape_Box_id_1_copy_0 [[CENTER=[159.178,101.793]] [POLYGON=[[147.664,90.307],[147.664,113.357],[170.714,113.357],[170.714,90.307]]]`
option 4: `EXCLUDE_OBJECT_DEFINE [NAME=Shape_Box_id_1_copy_0] [CENTER=159.178,101.793] [POLYGON=[[147.664,90.307],[147.664,113.357],[170.714,113.357],[170.714,90.307]]]`
option 5: `EXCLUDE_OBJECT_DEFINE [[NAME=Shape_Box_id_1_copy_0] [CENTER=159.178,101.793] [POLYGON=[[147.664,90.307],[147.664,113.357],[170.714,113.357],[170.714,90.307]]]]`

I have also tried to upload the same gcode manually, and automatically from PrusaSlicer, and nothing changed

I tried moving around the M117 command before the `START_PRINT` macro, but nothing. I have also tried deleting everything just before the `START_PRINT` (except for the object define).

Does anyone have any idea what on earth could it be? I have looked on google, Reddit, GitHub, Discord. No one has the same issue as me, there is nothing (or at least I couldn't find anything)

kyleisah, 2023-05-05:
What is M486? It looks like a macro that does something with excluded objects. Could this be it?

kyleisah, 2023-05-05:
[image]

kyleisah, 2023-07-06:
Try updating KAMP and make sure your slicer is defining objects correctly

closed 2023-07-06 by kyleisah