kyleisah/Klipper-Adaptive-Meshing-Purging #80: not compatible with beacon??

tnap1979, 2023-05-04:
i have a Beacon3d eddy current scanner and cannot get kamp working at all, am i missing something for beacon compatibility ?

tnap1979, 2023-05-04:
anyone ??

Carlibdanadr, 2023-05-04:
I'm currently having an issue too. I got adaptive line purge to work on one of my othe printers but with my current printer using the Beacon probe it isnt working, it does a full bed mesh and i dont get any purging. 

Im sure someone would help if you put some more information about the errors you're getting or what steps you have taken. 

Currently I am not getting  any errors is there a log somewhere in klipper that i can post for someone who needs to see?

tnap1979, 2023-05-05:
im not getting errors either, its doing a FULL bed scan and not an adaptive scan

Carlibdanadr, 2023-05-05:
I'm thinking it might have something to do with the fact that beacon replaces the [probe] section with [beacon] as stated in the documentation for beacon.

It's just a guess... I don't know anything lol.

kyleisah, 2023-05-05:
@tnap1979 Beacon uses it's own klipper module for leveling, homing, and generating a bed mesh, so I don't think it's compatible by default. I would have to have a beacon myself to figure out a way to make it work with KAMP unfortunately. I think you can still use the purging functionality though

closed 2023-05-05 by kyleisah

tnap1979, 2023-10-14:
im not sure whats gone on in the back ground,
but my beacon is now doing proper kamp scanning without issues
whatever happened, im thankful for :D

kyleisah, 2023-10-15:
@tnap1979 Fantastic, glad to hear. I think at first it didn't work and somewhere down the line the beacon.py module was updated and maybe that's what fixed it? Honestly have no idea. But I'm glad to hear you're KAMPing now. 😄