kyleisah/Klipper-Adaptive-Meshing-Purging #82: Internal error on command:"BED_MESH_CALIBRATE"

Darklordbigsword, 2023-05-04:
**Describe the issue**
When my Start G-code gets to the "BED_MESH_CALIBRATE" macro it spits out this error
Internal error on command:"START_PRINT"
Internal error on command:"BED_MESH_CALIBRATE"
Internal error on command:"_BED_MESH_CALIBRATE"

**What slicer are you using?**
SuperSlicer late 2.4 edition (Exclude object is fully functional.)
klipper v0.11.0-195-g3bd0be40
crowsnest v3.0.7-3-g20ed6a8
mainsail-config v1.0.0-11-ge4976f6
moonraker v0.8.0-39-g6c7dfe5
mainsail v2.5.1

**Troubleshooting**
I have tried changing the name of the macro along with its following parameters to not override the default bed mesh macro. 
I have checked and triple checked and from what i have found there is no other macro overriding the KAMP macro.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Klipper log file:**
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/11401977/klippy.log)

**Sliced Gcode**
https://github.com/Darklordbigsword/KAMP-issue-Gcode/blob/main/Calibration%20cube.gcode

**Additional context**
My current Start Gcode
https://github.com/Darklordbigsword/KAMP-issue-Gcode/blob/main/Start%20Marco

I myself do not have much experience with klipper code so my troubleshooting methods likely were improper and should not be trusted.
Thanks in advance for anyone whos willing to help!

kyleisah, 2023-05-05:
@Darklordbigsword Faulty region points are not supported, that is why the command keeps failing

Darklordbigsword, 2023-05-05:
Thanks for the quick response! Going to see if I can get it up and going now that I'm not making that mistake!

closed 2023-05-05 by Darklordbigsword